patience4711/read-APSystems-YC600-QS1-DS3 #89: new version

patience4711, 2023-02-22:
Right no we can only see inverterdata when we select an inverter in the frontpage. I am experimenting with a new version that has data from all inverters together in the frontpage. Please see the picture below to get an idea. 

Since i have no working ESP-ECU i can only test in a artificial environment, and that seems to work.  I would appreciate it very much if someone is willing to install this and give feedback.

You can upload the binary [here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7_exp.bin)

[image]

swbouman, 2023-02-22:
Next week, there should be 13 panels installed with 7 ds3 inverters. I’ll install this bin and test for you.

patience4711, 2023-02-22:
@swbouman  Great, thanks in advance.

Superbert25, 2023-02-26:
Hi,
I have installed this experimental version. I'm using 5 DS3L inverters with each 2 panels connected.

The version works fine upto now, I just noticed some strange behaviour in the build-up of the overview:

After initial access, only inverter 0 and inverter 1 are shown immediately:
[image]

After a couple of appprox 15 seconds, the table is completed:
[image]

Thanks for your effort in developing this solution!

R.

Eeg.

Superbert25, 2023-02-26:
My second screen-shot was from yesterday evening (inverters not yet paired), this one is from today:
[image]

patience4711, 2023-02-27:
@Superbert25 Thanks for your feedback, i solved that issue already. I think it looks good. Could you tell me how it looks on a smartphone?

Superbert25, 2023-02-27:
Sure! Screenshot from iPhone 13:

[image]

It looks fine to me :-)

R.  Eeg

patience4711, 2023-02-27:
Oke thanks, could you do this when there are values, preferably at the end of a shiny day, that will give me an idea.

Superbert25, 2023-02-28:
Would be good to truncate the amount of digits in the total. The amount of digits causes the table to shift outside the screen width (iPhone 13):

[image]

Based upon this screen-shot: If you truncate / round to 2 digits, I expect the table to fit the iPhone screen (even if total > 10 kWh total).

R.

Eeg.

patience4711, 2023-03-02:
I changed the format of the total energy value to 2 decimal digits. Here is the latest version 
[ESP-ECU-9_7_beta](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7_beta.bin)
 that also includes a feature to send mqtt output to a dedicated topic per inverter.

Superbert25, 2023-03-02:
I am now using the second 9.6 experimental version. Couple of times a day the Zigbee Coordinator is restarted, sometimes the system as a whole is rebooted. I noticed that it seems related to loosing communication with one or more inverters.

I will build a new setup in a couple of weeks time with a longe range zigbee module (with PA and external antenna, that might improve things. I can share my findings here if you like.

Superbert25, 2023-03-02:
Just installed the 9.7 beta:
- Issue of showing only 2 inverters at accessing the home screen is back in again
- Total is not limited to 2 digits
- 2 Reboots shortly after each other after installation

[image]

[image]

Superbert25, 2023-03-02:
> I am now using the second 9.6 experimental version. Couple of times a day the Zigbee Coordinator is restarted, sometimes the system as a whole is rebooted. I noticed that it seems related to loosing [...]

My bad! I was actually using ESP-ECU_v9_7_exp_exp.ino.nodemcu.

Now reverting back to this experimental release as the beta is having some issues.

patience4711, 2023-03-02:
This is all a little confusing, what i am trying to find out is if the version [9_6b](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_6b.bin) is working stable and the [9_7_beta](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7_beta.bin) is not stable in your situation.  Otherwise i have no clou where to find the pain in the program. These are freshly compiled files, the 9_6b is 9_6 with the mqtt topic feature. The 9_7_beta has above that the new frontpage. 

Indeed,  the cc25xx crashes when there is bad communication with an inverter, this happens a lot when a high poll frequency is used. When the esp itself crashes than it could be a watchdogtimer issue.

Superbert25, 2023-03-02:
I will install 9_6b now, but can only say tomorrow if that version is working fine (for me). If time permits, I will try 9_7_beta in the afternoon (next week I'm not able to perform any testing, so could be that you won't get any feedback from me until the week thereafter).

R.

Eeg.

frtz13, 2023-03-02:
I installed the 9_7_beta an hour ago. until now, everything works as expected. I will report back tomorrow evening.

patience4711, 2023-03-03:
@frtz13 Good to hear that. It would be helpfull if you could tell us in your next report:
- what CC25xx you use? 
- How many inverters/type you have
- do you experience crashes of the esp or the cc25xx
- does the energy total have 2 decimal digits? Frontpage on smartphone oke?

swbouman, 2023-03-03:
can anyone help in understanding the output values? I assume the power value is the current power in Watt per panel, but when I multiply dcv*dcc the math has a different answer. How do I have to interpret the power and energy value?

                    dcv. dcc.  pow(w) energy
    "ch0": [ 32.2, 9.6,     274.7,   579.11]

The system is still crashing with 7 inverters polled. Ver (9.6b) 

I use a CC2530 + 2591
7 inverters ds3
And the totals look fine, but have a lot of digits sometimes.

patience4711, 2023-03-03:
When i downloaded and installed version 9_6b  a minute ago i get this frontpage. You?
[image]

Power is calculated as follows ( (energy poll x+1) - energy poll x) ) / 300 sec )  So power is the avarage power between 2 polls. It is obvious that you can't use the u*i formula since these are momentary values that fluctuate a lot within 5 minutes.

Superbert25, 2023-03-03:
I’m new, started with 9.6, now on 9.6b (again).

9.7 beta crashed all the time (within 30 seconds). I flashed over OTA, could that have something to do with it? So, I reverted back to 9.6b over OTA, which is stable enough (few crashes a day). I can dive into the issues after next week.

R.  Eeg.

frtz13, 2023-03-03:
my report about the 9_7_beta:
I have 2 YC600 inverters, approx. 35 % resp. 50% Zigbee signal strength.
the ESP-ECU ran without problems for 25h, no crash,
resetCounter = 1.
CC2530 zigbee module.
Frontpage looks ok on my phone. I never saw more than 2 decimals.
works fine ! thanks.

swbouman, 2023-03-03:
@patience4711 , thank you for your response! Your answer is clear to me, the polled data can be a few minutes old and therefor not really accurate. 

About the crashes: the software is stable during night time. During polling time the esp crashes a lot. I’ll try to power the cc2530/2591 direct from a power source. The behavior looks like a power shortage. The 2530 is polling and the esp is sending data over WiFi. That consumes a lot of energy. 

I’ll build the new power source tomorrow and report back to you.

patience4711, 2023-03-03:
@swbouman  You still didn't answer my question about the frontpage of 9_6b. Since you where mentioning the totals (not present in 6b) I thought maybe you have the wrong version.  If the problem is related to the number of inverters you should be able to find a stable situation by lowering the number of inverters.
If your powersupply is 3A there shouldn't be problems but if not there could be brownouts. 

@Superbert25 What zigbee do you use, by chance the 2530+2591?

swbouman, 2023-03-03:
@patience4711 at the moment I’m running 9.6b. The program is stable since night mode. 
that’s why my thought are about the power consumption of the wemos and the zigbee unit. 

I’m currently powering the unit with a 2 amps usb adapter. So that could be the problem. I keep you posted. 

Next week the installer will check the inverters on the roof. The polling data gives 11vac on most of the roof inverters. No power is produced. 

Here is my screenshot.

joffito, 2023-03-03:
Maybe a big cap like 1000uF across the ESP power pins is enough to keep it running. 2 A sounds like enough to me.
How is the DS3L wired up. Each Inverter goes to the grid separately (or via a node) or are they kind of daisy chained / pass through system ?

patience4711, 2023-03-03:
If an esp can draw 400mA than this is 0.4*5=2A but when a hungry cc2530+2951 is connected . . . . . 
In the building instructions a 3A powersupply is recommended.
To rule out all of the new features, please test with 9_6a, this has been proven to work !!!!!

joffito, 2023-03-04:
i think it only pulls the 400mA very briefly and also at 3.3V. so with a big cap that could be buffered. but i don't know how much the ZB module pulls. anyhow i am curious if the roor cause can be found. having a decoupling cap certainly does not make things worse, also not with a 3 A device as the peal current may be the culprit

Superbert25, 2023-03-04:
> 

I am using a cheap CC2530 without amplifier. I ordered a new one with 2592 and external antenna connection.

patience4711, 2023-03-04:
@joffito  Do you use that big cap yourself? I am sure that you don't need it if you have a good 3A powersupply. 
Could you tell us what version you are running with what hardware/inverter(s) and if there are stability problems?

joffito, 2023-03-04:
No i don't use it with my simple CC2530 and ESP with only 1 DS3S inverter paired. But it helped in the past with some GSM modules. But you're probably right - a good 3A supply should be sufficient. Let's see how it turns out. I am running the 'original' 9.6. very stable since 11 d. it would also be interesting if people with trouble wired or soldered the zigbee module to the ESP

frtz13, 2023-03-05:
ESP-ECU v. 9_7_beta up for 3 days. resetCounter = 8.
D1-mini pro board + cc2530 running from a 5V / 1A power supply.

patience4711, 2023-03-06:
@frtz13  thanks, so the esp stays stable meaning that the new features are not involved. We can see here however that the zigbee crashes some times. This is an old problem that occurred with a high polling frequency (more than once in 5 minutes). In the past the polling frequency was configurable but since it didn't work stable the 5 minutes (same as the official ecu) became standard. Multiple inverters means virtually a higher polling rate and that causes the crashes of the zigbee. The reason is supposedly hidden in the zigbee's software (a buffer overrun?) but the designer is convinced that this is not the case. We know that when this happens very large serial messages are sent. This also could cause the zigbee to crash because the reading of the serial buffer takes too long.  Maybe the wdt overflows because of that. I really don't know, 

So what can I do? I am thinking of some sanity check and buffercleaning in between the polling. Maybe this could help and it certainly won't harm. I will work this out and will come up with a modified version.

swbouman, 2023-03-06:
Hello all,

After a weekend of testing with a 10 A power supply, just to be sure 😜, and a lot of crashes when switching from the console view to the home view, I installed the new 9.7 version today. 
This latest version gives me a solid online system for 2 hours now. 
I got some error 50 status codes, but after a minute or 10 the system restarts the zigbee module. Mem is around 16000 and stable. 

To avoid the error 50 code, a little delay between polling different inverters could be the(or a) solution.

patience4711, 2023-03-06:
Oke so now the situation is that the esp is stable but the zigbee still crashes. I am thinking about a method that divides 5 minutes by the number of inverters and uses this result as a pause between pollings of each inverter. You may be right that this could help but i remember that every higher polling rate than 5 minutes leaded to crashes of the zigbee. Nevertheless i'l give it a shot. I'l come back when i have this implemented.

swbouman, 2023-03-06:
@patience4711 , you’re right about the stablility. Now solid for 3 hours. 

About the zigbee stability, could you pls start with a delay of 2 seconds between polling each inverter, everytime auto polling is started? So the polling is started after 300 secs, with a timeout of 2 secs after an inverter poll is started? In my situation the polling of all inverters will take 14 secs. I think this is a minor code change for your program.

patience4711, 2023-03-06:
Oke [here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7xb.bin) it is. With a delay of 2 seconds between each poll. Can you explain why you think that this would make a difference? I hope you are right but wit the story of the high poll rate in mind...

swbouman, 2023-03-06:
It is a shot in the wild @patience4711. I took the higher polling rate in mind. I’m the console I saw the polled inverters coming in quickly after each other. In mine opinion a new big call can be written to the cc2530 wile it is still processing the previous. So that’s why I think this could be a solution. 

An other thing I think about is the disabled serial debugging from the esp when zigbee is  enabled. I don’t know if the serial debugging from the esp is disabled at that moment or if it is forwarded to the cc2530. That could cause a failure in the cc2530 due to not handled serial commands… that would make a big investigation. 

I’ll test the new firmware tomorrow. Thanks for the big support to this project!

swbouman, 2023-03-07:
Hello,

I tested the new 9.7 firmware today. Uptime is for about 4 hours, where after the esp reboots. I’ll investigate if something has todo with the WiFi, but I doubt that because all other esp’s are running stable for months. 

Ive not experienced any error 50s anymore. So the break in polling between inverters is an outcome. 

Ive another strange behavior. Two inverters (far away from the esp) have polling issues with code 11 and 15. At the next polling they give data and a signal strength about 30% 

Does anyone has this issue aswell? When auto polling is off, and inverters are polled by sending the poll command over mqtt polling is instant for these two inverters. 

[image]

—-

[image]

swbouman, 2023-03-07:
[image]

swbouman, 2023-03-07:
[image]

patience4711, 2023-03-07:
That ESP crash after some hours,.. weird.  I am still wondering if this has something to do with the number of inverters. The other issues seem to be a range problem to me. 
At this point i wouldn't know what else to do. If the other users share their experiences with this [9_7xb](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7xb.bin) version here we might find a systematical  reason for the crashes.

frtz13, 2023-03-07:
installed the 9_7xb version. I'll report back tomorrow.

frtz13, 2023-03-08:
version 9_7xb up for 11h now, no problems. resetCounter = 0.
on thing I would like to mention: I never stay connected to the ESP-ECU web interface with the browser. I just have a look and close the window afterwards. Especially, the INFO page makes http requests continuously, retrieving the current time.

patience4711, 2023-03-08:
@swbouman as far as i can see only inverters 0, 3, 6 show a correct polling answer. This is weird because the program checks some properties of the answer and rejects it if it not matches the criteria. 
Did you read [here](https://github.com/patience4711/RPI-APS-inverters/issues/77) that there seems to be a batch of ds3 inverters that have issues?

swbouman, 2023-03-08:
@patience4711 thank you for looking in the data. As far as I know, inv3 has one panel attached, so I’ll disable the second one in the near future. 

Today the installers are back because of the wrong readings. They discovered that one inverter is probably dead because of bad installation and the others weren’t attached to 220v grid at all. 

This was in the morning and they went shopping for tools since then. 

Hope they will be back today to finish the job. Then all inverters will produce power. 

The company owner was here to apologize for the first group of installers, which worked here.

patience4711, 2023-03-08:
@swbouman this is a scandal, not much can go wrong with the installation, bunglers! Anyway now there is hope that the ECU can work properly.

frtz13, 2023-03-08:
@patience4711 I connected to the ESP_ECU with 3 browser windows. navigated with one of these to the INFO page. the ESP-ECU crashed rapidly, as soon as I started to navigate.
while the front page is relatively sparse with HTTP requests (about 1 for each inverter every 10 seconds approx.), the INFO page makes approx. 10 requests per second to get the current time.

patience4711, 2023-03-08:
@frtz13 Thanks for testing, if i open 4 browserpages (3 on chrome and one on my phone, all to the infopage, no crashes occure. But i don't have traffic on the frontpage so it is not completly comparable. The responsiveness decreases indeed. The refreshrate of the time is about 2 seconds, i can change in such way that the script only runs once at pageload. It is only a check that system time is correct so not really important.
[v9_7xc](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7xc.bin)

frtz13, 2023-03-08:
in the 9_7xb version, INFO page, the HTTP request rate for the current time is much greater than every 2 seconds. it is about 10 per second, maybe more. checked with both Firefox and Edge (Chromium).
ok, in the 9_7xc version, these request are gone.
BTW, would be nice to have a /get.rssi request to get the current wifi signal level.

patience4711, 2023-03-08:
The wifi signal strength is displayed in the infopage. If you refresh the page you'l get the current level.  ( WiFi.RSSI()) )

frtz13, 2023-03-08:
... I know. the idea is to request is once a minute or so, and display it in my home automation system.

patience4711, 2023-03-09:
Sorry too less programspace left for "nice to have" additions. In the childhood of this project i had this to monitor the free memory to see if the crashes could be explained with it. Lets concentrate on getting the system stable insteat.

frtz13, 2023-03-09:
9_7xc version up for 18h. no crash, resetCounter = 0.

frtz13, 2023-03-10:
9_7xc version up for 48h. no crash. resetCounter = 0, which is significantly less than previous versions. well done, @patience4711 .

frtz13, 2023-03-12:
9_7xc version: got a reboot this morning, when connecting to the web interface. It looked like the ESP choked when fetching the data for the second inverter. I also got a Zigbee reset this afternoon.

patience4711, 2023-03-13:
@frtz13  You can see the difficulty here, these incidents don't happen systematically so it's impossible to find out what the cause is. If it is the websever that crashes than it must be the new frontpage causing that. Because to my knowledge this doesn't happen with the former versions. Would be interesting to know how if the 9_6 version works stable in your situation.
But if it is the frontpage why not everytime you visit it?  

As for the zigbee, the cc2530 is seems pretty outdated and its memory is vulnerable for corruption, Zigbee2Mqtt doesn't recomment it for this reason. There is however no alternative that has the right firmware and can be connected to the ESP.  After a lot of experiments i couldn't find a possible reason (could be firmware related) so i made the system invulnerable for these crashes. When it happens, every 10 minutes there is a healthcheck that recoveres the zigbee. In worst case you could miss 2 polls. We'll have to live with it I'm afraid.

frtz13, 2023-03-13:
>> But if it is the frontpage why not everytime you visit it?
Maybe ist just concurrence of events. display the frontpage while zigbee traffic going on...
I hardly ever display the ESP-ECU web pages. Mostly to check the wifi rssi from time to time (the ESP-ECU is quite far away in the garden). And I did more often recently, to give some feedback about the new version.
the 9_6 version had been running for months without problems. just a Zigbee reset from time to time.

patience4711, 2023-03-14:
@frtz13 So it must be the new frontpage then. Requesting data from all inverters at the same time may be too much for the server. Could even be that the mosquitto network activities cause it. I don't know, the system has an async webserver that runs its business in the background. It is very hard to debug.

So for you the new frontpage is not very important an the old version is goed enough?

I was wondering: Sometimes it seems that something is stacking up in the zigbeemodule, that would explain why the more polls id does, the sooner it crashes. If we would force a reset of the ZBmodude every midnight, than its memory is cleared and maybe is wouldn't crash during the day. So i compiled [v9_7xd](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7xd.bin) with this idea implemented.

patience4711, 2023-03-14:
@frtz13  Yes i forgot that, sorry. Could you download and install it again [v9_7xd](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7xd.bin) , i made another minor change to the significance of the resetcounter, These forced resets do not increase the resetcounter, only the incidental resets are counted. So you could see in the log that the reset took place but the resetcounter is still 0 (hopefully).

frtz13, 2023-03-14:
@patience4711 ok, done.

frtz13, 2023-03-16:
v.9_7xd. uptime: 2 days. resetCounter=1.

[40 more replies]

closed 2023-04-14 by patience4711