patience4711/read-APSystems-YC600-QS1-DS3 #90: Feature Request: MQTT

Superbert25, 2023-02-26:
Not really an issue, but I think it would be nice to have the ability to publish the results of the individual inverters to separate MQTT topics.

Suggestion: Use the inverter ID and/or name as an MQTT topic suffix --> Example:

Inverter 0
Name: example
send topic: ECU

Publish to: ECU/example0

swbouman, 2023-02-27:
I'd like to have this Pull request in the software. In the current situation the data splitting asks for +/- 50 lines of code to set all the sensors of 1 intverter. In my case with 7 inverters the config will be a mess. If the data is initially splitted, the config will be compact. 

Hope you could help @patience4711

Superbert25, 2023-02-27:
Perfect!

frtz13, 2023-02-27:
thanks, @patience4711. I installed the binary from above. configured the send topic with a slash at the end. When I click the "test" button, the following happens:
- a mqtt message gets sent to the (send topic)/110 topic.
- the mqtt send topic on  the mosquitto configuration page is changed to (send topic)/110 permanently.
this does not look good.
FYI I use format 3.
not we are after sunset here, so I wait for tomorrow. however, I have the impression that even inverter messages are sent to the 110 topic, too. for now, I reconfigured the send topic with the slash at the end. I'll report back tomorrow.
BTW: If I configure a send topic without a slash at the end, the test message is sent to the (send topic)/pv topic and the send topic is not changed in the config.
BTW 2: the home page looks fine on my phone.

patience4711, 2023-02-27:
Indeed the test changed the value of topic. And the new value didn't have the slash anymore so it was not recognized as a "special" value. My mistake. This is corrected now, try [this](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_7_exp_exp.ino.nodemcu.bin)

Do you think that when the value of the total in the homepage exceeds 10, it still would look oke on a smartphone? Even better, could you post a screenshot.

frtz13, 2023-02-27:
thanks @patience4711 . this looks better. I'll report back tomorrow.
sorry, I will not have a production over 10kWh (I only have 2 inverters).

frtz13, 2023-02-28:
I configured the send_topic as ending with a slash. now the mqtt messages for both of my inverters are sent to the (send_topic)/110 topic, and _not_ to different topics for each inverter.

frtz13, 2023-02-28:
I also had a spontaneous reboot of the ECU, which did not happen with the 9_6 version, which had been running for months.

swbouman, 2023-03-01:
Here it is working fine. I defined the IDX in the inverter setup. Every inverter has a number, inverter 0 has idx 1, inverter 1 has idx 2. I'm posting to topic homeassistant/inverter/ and I defined sensors in my config. Data is coming through. 

At the moment the other panels are installed so I hope I can report the other inverters soon. 

@patience4711 thank you so much. For your own convenience it will be helpful if you report your library names and versions. If others can compile the sketch, we can help in further development.

frtz13, 2023-03-01:
thanks @swbouman to remind me of the DOM.IDX parameter, which I had forgotten about (with previous versions it did not seem to have any function).
Now this works as expected, and I get MQTT messages to seperate topics.
@patience4711 any chance to get an option to send these MQTT messages with the retain-flag set ? Maybe use DOM.IDX values >10, or something?

swbouman, 2023-03-01:
hello all! 

My panels just got installed. As you can see in the mobile screenshot, energy above 10 KWH are displayed correct. 

I have serious problems with pairing and polling. When pairing, random answers are connected to the inverters. So I have more inverters (other serials) with the same pair code. 

System is crashing more times an hour. 

[image]

patience4711, 2023-03-02:
Conclusion: if we add a slash to the output topic, (Mqtt settings) the Dom idx (inverter settings) is added to the topic.

p.e. We defined output topic **"john/garreth/"** than the output of an inverter with idx **"123"** is sent to the topic **"john/garreth/123"**

closed 2023-03-02 by patience4711

Superbert25, 2023-03-02:
Works fine in the second 9.6 experimental version. Thanks! I Will check the new version soon