patience4711/read-APSystems-YC600-QS1-DS3 #93: First Wifi connection via hotspot

skizzo3000, 2023-03-14:
Hi,
i'm trying to use NodeMCU v3 with ESP-ECU_v9_7xd.bin or older (ESP-ECU_v9_6b.bin, ESP-ECU_v9_7xc.bin) but after successfully setup wifi network via hotspot (page shown IP) the module restart and enter in scan start every time.
Serial log below 

```
/wifiCon requested
send confirm page  
yoepy, connected
portal timed out, resetting...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
   scan start
15 networks found
```

In the repo there aren't the last source code so i cannot investigate the problem.
Anyone can help me to makes it working ?

Thanks
Sk3

patience4711, 2023-03-15:
I quote from an old issue: did you try to completely erase the flash memory (using esptools, for ex.), before flashing the ESP again?

skizzo3000, 2023-03-15:
@patience4711 Thanks for your support.
I erased twice via esptools and now it's Works !

P.S. Can you update the ino files to 9.7xd version ?
Thanks
Sk3

patience4711, 2023-03-16:
@skizzo3000  When it is clear that this version works without problems. You can contribute by sharing your experiences. Please mention your zigbee module and how many/type of inverters.

skizzo3000, 2023-03-17:
@patience4711 Sure !
My inverters are on the way !!

skizzo3000, 2023-03-26:
@patience4711 after several try to paring inverter, was paired correctly. DS3L-EU
How can i make polling every minute or less ?
Thanks
Sk3

joffito, 2023-03-26:
Read the wiki. Polling more frequently than 5 min needs to be done manually.

But it has been discussed a 100 times that this will lead to instability due to the ZigBee. also APSystems does not poll more often than every 5 min.

skizzo3000, 2023-03-26:
@joffito thanks ! Ok i will setup a measurement by external power meter.

closed 2023-03-26 by skizzo3000