patience4711/read-APSystems-YC600-QS1-DS3 #94: Issue with DS3-L data reading?

LordDarkneo, 2023-03-24:
Hello,
I finally managed to find some time to test the overall installation with my DS3-L with 2 solar panels connected on it.

What's strange is that I am having the correct pairing / data from the info page:
[image]

But the production remains at 0.0....
[image]

Any idea on how to check / verify that everything's OK? 

Thanks for your help

joffito, 2023-03-24:
Are you sure that your inverter is connected to mains properly (maybe switch L and N) ?
Can you verify somehow that is is not damaged due to some issue like overvoltage, wrong installation ?

in another issue someone also had trouble with the inverter and the AC value was low. turned out the company did not install the inverters correctly.

frtz13, 2023-03-24:
@LordDarkneo did you check this page about DS3L inverters having firmware problems?
https://www.quesolar.com/apsystems-storing-ds3-en-ds3-l-oktober-november-2022/
according to the visible part of the serial number, your inverter may need a firmware update.

LordDarkneo, 2023-03-24:
@frtz13 *** it... When I received mine, it was supposed to have no production issue... You're right, my SN indicates that I need to update the firmware...
@joffito Inverting L and N is not possible on DS3L, has connectors are différent (you cannot plug the L on the N, that's just mecanically impossible :) )

patience4711, 2023-03-24:
Moreover we have the console for debug purposes. We can manually poll the inverter and observe the answer. Start reading the wiki!

LordDarkneo, 2023-03-25:
@patience4711 I have been too enthusiastic... I just plugged one solar panel (uppon the 2 I have) just to test the ESP ECU (pairing and info page). But I did not get too much into details, as the info was correct (even if the production wasn't displayed, I thought is was because of the cloudy winter day...).
Anyway, the DS3L has been returned back to seller for update, I'll perform 10;DIAG and 10;pool=0 once I get it back if it's still not working

swbouman, 2023-03-25:
In my opinion the acv line was not connected, nor there is ac voltage on the line. Please check that first. 

In my installation this caused your output.

closed 2023-04-14 by patience4711

LordDarkneo, 2023-04-14:
> In my opinion the acv line was not connected, nor there is ac voltage on the line. Please check that first.
> 
> In my installation this caused your output.

I think the DS3-L was malfunctionning. I sent it back to the seller, and still waiting for his feedback. Anyhow, the pairing was still ok, I was able to see the DC volts and send the 10;polling=0 (and yes, the inverter was correctly connected to an AC line ;) )

The issue is (for my point of view) clearly not related to the ESP ECU, asI was able to get the information from the inverter....

LordDarkneo, 2023-04-19:
Quick update (in case someone is having the same issue). the reseller finally changed my DS3-L (to a 703000547xxx serial number). After plugging one solar panelto perform some tests, everything seems to be finally OK!
[image]
That's just after pairing (what's strange is the solar panel having 0.1v when there is nothing connected to it). After few hours, here's what I have in domoticz:
[image]
So now they can be moved to their final location. Thanks for all (and I confirm the DS3-L support then 👍 )