patience4711/read-APSystems-YC600-QS1-DS3 #97: Zigbee crashes

togoright, 2023-04-01:
Hello patience4711,

I want to thank you for the great job you did. I've been using your program since the end of 2021 and I'm very excited.

Before I knew your project, I had neither programmed in C++ nor compiled a program, so it was completely new territory for me. In the meantime, I have made many changes to adapt it to my needs. I probably also found a solution for the hang-ups when querying the inverter data too often. I've been polling every two minutes since the beginning. In the beginning I also had the hangers again and again. Since I create the pollcommand differently, I don't have them anymore. As an experiment, I polled every 10 seconds for two days without any problems. Even with 5 seconds there were no hangups on two days, but there was a one-time error when decoding or calculating the total daily yield. I didn't pursue this any further because I switched back to a polling time of 2 minutes.
I have 1 YC600 and use a D1-mini pro with CC2530+CC2591.

Here my change in ZIGBEE_POLLING.ino:

`
//    char pollCommand[254] = {0};
//    char normalOperationBaseCommand[][254]  =
//    {
//        "2401",                      //+ ..string.sub(inv_id[x],3,4)..string.sub(inv_id[x],1,2)..
//        "1414060001000F13",          //append (concatenate) _ecu_id_reverse
//         // "1414000601000F13",          //append (concatenate) _ecu_id_reverse
//        "FBFB06BB000000000000C1FEFE" // end of String
//    };
//    
//    //Serial.println("Inv_Prop[which].invID = " + String(Inv_Prop[which].invID) );
//    char inv_id[7];
//    memset(&inv_id[0], 0, sizeof(inv_id)); //zero out otherwise we get strange tokens
//    delayMicroseconds(250);
//    
//    strncpy(inv_id, Inv_Prop[which].invID, strlen(Inv_Prop[which].invID));
//    inv_id[7]='\0'; //terminate
//
//    char ecu_id_reverse[13];  
//    ECU_REVERSE().toCharArray(ecu_id_reverse, 13);
//    
//    strncpy(pollCommand, normalOperationBaseCommand[0], strlen(normalOperationBaseCommand[0])); // the 2401
//    //Serial.println("pollCommand now = " + String(pollCommand));
//    strncat(pollCommand, inv_id + 4, 2 ); // ad the 2nd byte of inv_id
//    strncat(pollCommand, inv_id + 2, 2 );     // ad the 1st byte of inv_id
//    //Serial.println("pollCommand after reversed inv id = " + String(pollCommand));
//    strncat(pollCommand, normalOperationBaseCommand[1], sizeof(normalOperationBaseCommand[1]));
//    //Serial.println("pollCommand after 1414060001000F13 = " + String(pollCommand));         
//    strncat(pollCommand, ecu_id_reverse, sizeof(ecu_id_reverse));
//    //Serial.println("pollCommand after reversed in id = " + String(pollCommand));
//    strncat(pollCommand, normalOperationBaseCommand[2], sizeof(normalOperationBaseCommand[2]));
//    //Serial.println("pollCommand after FBFB06BB000000000000C1FEFE = " + String(pollCommand));
//    
//    // calculate length and put this at beginning
//    strcpy(pollCommand, strncat(sLen(pollCommand), pollCommand, sizeof(sLen(pollCommand)) + sizeof(pollCommand))); //build command plus sln at the beginning
//    // put in the CRC at the end of the command
//    strcpy(pollCommand, strncat(pollCommand, checkSum(pollCommand), sizeof(pollCommand) + sizeof(checkSum(pollCommand))));
//    //DebugPrintln("pollCommand:");  //DebugPrintln(String(pollCommand));

    char pollCommand[67] = {0};
    char ecu_id_reverse[13];  
    ECU_REVERSE().toCharArray(ecu_id_reverse, 13);

    snprintf(pollCommand, sizeof(pollCommand), "1D2401%.2s%.2s1414060001000F13%sFBFB06BB000000000000C1FEFE", Inv_Prop[which].invID + 4, Inv_Prop[which].invID + 2, ecu_id_reverse);

    // put in the CRC at the end of the command
    strncat(pollCommand, checkSum(pollCommand), 2);

    if(diagNose) ws.textAll("zb send poll cmd inverter " + String(which) + "  cmd:" + String(pollCommand));   
        #ifdef DEBUG
        swap_to_zb(); // set serial to zb  
        #endif

    sendZigbee(pollCommand);
`

To relieve the processor, it might be beneficial to create the ecu_id_reverse once and store it in a variable so that it doesn't have to be recreated each time.

Best regards

patience4711, 2023-04-01:
@togoright Hi thanks for this contribution. It would be good news that this would help to avoid the zigbee crashes. 
Your solution seems a lot simpler but i cant I can't oversee if this is actually the case. Therefor my knowledge is not sufficient, i am not a programmer at all you see. But the benefit as to the memory is obvious.

I see that the data length is hardcoded now, (1D). You  assumed that the only variables are the inv-id and the ecu_id, so that the command it has always the same length.
I noticed that because the length addition **strcpy(pollCommand, strncat(sLen(pollCommand), pollCommand, sizeof(sLen(pollCommand)) + sizeof(pollCommand)));**  caused a crash in the esp32. So i considered to hard-code that too. I found a workaround for this but here is something going on that you avoided in your code.
I am going to do some tests with this and when it works i'l include it in the software.
Thanks again, really appreciate it.

togoright, 2023-04-01:
Wow, you are very fast 👍

I'm not a programmer either, I only do it as a hobby.

swbouman, 2023-04-01:
I installed the XF version, uptime is 32 min, free mem: 16632 bytes-- with 7 inverters, reset counter is 3.

I keep you posted.

patience4711, 2023-04-04:
@togoright  yes, maybe too fast. I commented out my polling function and replaced it with yours. What happens now is that at every healtcheck the ping to the zigbee (and other communication) fails. Meaning that the coordinator will be restarted. 
When i disable the autopolling this doesn't happen. I can poll via the console and do healtchecks as much as i like, the coordinator remains up.  If the problem has something to do with your polling code, the problem should also occure when manually polling, one should think.
When i reverse the polling methods back, this behaviour is gone. 
I tried some other things, changed the reset pin for instance, made the reset permanently high, nothing helped. this is really driving me nuts.
The healthcheck is once in 10 minutes. When autopolling i can do healthchecks in between, all oke but when the sceduled healthcheck takes place, it goes wrong.

In short, after some other tests the situation with your code is this:

- When auto polled, the healthcheck fails all the time.
- manually heathchecks do not fail. 
- When not auto but manally polled, the healthcheck doesn't fail.
- When no auto healthcheck, the coordinator keeps working.
- With the original polling code there are no problems.

So what I did is skipping the automatic healthcheck and set a flag for the healtcheck when the length of a poll answer is 0. But that is not a real solution.

The reason i am telling you this is that maybe you encountered also some strange behaviour while implementing your code.

patience4711, 2023-04-04:
After a lot of headbanging i figured that with autopolling/healtcheck they coincided. This is not the case when we do this manually. Could that have something to do with it? But why not with the original code? Anyway, the healthcheck was every 10 minutes and now 12 minutes. Problem seems gone.

joffito, 2023-04-04:
with 12 min for healtcheck wouldn't polling and health check collide again after 5 'rounds'? better use 13 min. but you're probably on the right track! there ought to be a pause between both operations that give the zigbee module enough time to process. question is how long that pause needs to be.

patience4711, 2023-04-04:
Yeah, maybe better no automatic healtcheck at all but put a flag up when a polling answer == 0, this indicates that the coordinator may be down. 
thanks for thinking along.

swbouman, 2023-04-04:
Could it be an option to build in a polling check. If polling of a inverter fail, retry polling that inverter. if the second poll fails, do a health check and restart the polling sequence? 

I have two inverters mounted on a longer distance. Sometimes polling fails, but after a second try the polling of the inverter succeeds. 

I dont have anby zigbee crashes anymore after installing the G version. Hence the frem mem is 13000 bytes after 2 days uptime. I still believe som mem is beeing held by the program (webinterface pages).

And still the ecu is rebooting after closing the console page by pressing done. Hope you could look into that.

togoright, 2023-04-04:
> @togoright yes, maybe too fast. ...

I'm surprised it causes you such problems.
Before I wrote the post, to be on the safe side, I downloaded an up-to-date copy of the project and made the changes described. I also made the pollres adjustable, added my format for MQTT and set the refresh time of the info page to 2000ms (otherwise it is reloaded every 100ms for me, which often leads to reboots). I tested this version with my inverter for two days without any problems.
Unfortunately I can't make this version available to you, but I will make the changes again, compile it and post it here with the source code. At the same time, I will also flash the version for myself. I don't have much time at the moment, so it will probably take a while.

patience4711, 2023-04-04:
That's oke, now i do the healtcheck with some offset relative to the polling. Problem seems solved.

patience4711, 2023-04-04:
@swbouman  the subject here is zigbee crashes. Could you move your posts to the new version issues. This way we keep the issues mored readable. Thanks

closed 2023-04-06 by patience4711