patience4711/read-APSystems-YC600-QS1-DS3 #98: Problem pairing DS3 Inverter properly

JanMueller207, 2023-04-01:
Hello everyone,

and first of all thank you for sharing this project, good work!
Anyway, I tried to pair my AP DS3-S to the Node MCU with Zigbee Module and after some trials the inverter was successfully paired and is listed in the overview. But it is not displaying any value (just n/a) and it always displays "Initializing the Zigbee Radio..." in the first tab.

The communication between NodeMCU and Zigbee seems to work fine, I tested it according to "To test the zigbee hardware you can type "10;ZBT=2710AABBCC" in the console. You should get inMessage: FE036710AABBCCA9"

[image]

Any Ideas well appreciated, thanks in advance. :-)

joffito, 2023-04-01:
Can you do a health check of the zigbee via the console?

joffito, 2023-04-02:
10;health

JanMueller207, 2023-04-02:
Thank you for the Quick reply, here are 2 answers from the Console:
I thought the connection is okay due to the answer...
[image]
[image]

joffito, 2023-04-02:
which zigbee module and firmware are you using?

joffito, 2023-04-03:
Did you erase the chip prior to flashing? i think somebody reported that he had to flash several times before it worked. maybe also just try the .bin file. 
otherwise i don't know what could be the cause. i used esptools under linux and it worked well but i had to erase the flash first. 
also double check the wiring to the esp ;-)

patience4711, 2023-04-03:
To me it seems that the wiring is oke, the loopbacktest works. The coordinator doesn't get up. What version of the ESP-ECU did you install?

JanMueller207, 2023-04-04:
> Did you erase the chip prior to flashing? i think somebody reported that he had to flash several times before it worked. maybe also just try the .bin file.
> otherwise i don't know what could be the [...]

I did not erase the Chip. Will try it out next time, is there an Option with CCLoader?
Can I find the final bin file for CCLoader somewhere in the folder? Would like to make Sure, that the compilation was fine.

JanMueller207, 2023-04-04:
> To me it seems that the wiring is oke, the loopbacktest works so the serial communication is oke. The coordinator doesn't get up, thats the real problem here. 
> Could be the reset wire, wrong firmware or ... since you seem to have a self compiled version of the ESP ECU....?

I uploaded the ESP ECU file from your folder from Arduino and Compiled it there. So it should be the original one or where are your doubts coming from?
Thank you!

joffito, 2023-04-04:
> I did not erase the Chip. Will try it out next time, is there an Option with CCLoader? Can I find the final bin file for CCLoader somewhere in the folder? Would like to make Sure, that the [...]

I never used CCLoader, sorry. I used a Raspberry Pi and this tutorial: https://blog.boris-wach.de/permalink/265. I think i just had to get WiringPi from another source.

1. Erase the ZigBee Flash
2. Flash directly the Hex-File provided (no need for any recompilations)

And just to try it out: flash the .bin file for the ESP directly which is provided like 9_6c. No own compiling.

OR: your ZigBee module is faulty...

patience4711, 2023-04-04:
@JanMueller207 I never saw any german laguage in my console so i thought you must have your own compilation of ESP-ECU.

JanMueller207, 2023-04-04:
> I never used CCLoader, sorry. I used a Raspberry Pi and this tutorial: https://blog.boris-wach.de/permalink/265. I think i just had to get WiringPi from another source.
>
> 1. Erase the ZigBee Flash
> 2. [...]

Thank you, will try that next time. On the first view I did not find the provided hex file in the files here, Could you pls tell me where exactly!?

closed 2023-04-20 by patience4711

JanMueller207, 2023-05-21:
Hello Patience 4711,

I uploaded the newest FW to the ESP and now the Coordinator seems so start:
[image]

The inverter is not here so I cannot test the pairing but it seems to work so far in between the ESP ECU <-> Zigbee module!?