patience4711/read-APSystems-YC600-QS1-DS3 #99: MQTT en_01 / en_02 incorrect (v9_8)

Superbert25, 2023-04-05:
This issue is related to version 9_8:

MQTT reports faulty values for en_01 and en_02. Screenshot ECU server:
[image]

Screenshot from my automation server:
[image]

Power values are matching, energy is completely wrong (was earlier today fine). Could have to do with exceeding some value?

R.

Eeg.

Superbert25, 2023-04-05:
After a reboot of the ECU, the webserver was also displaying the faulty values...maybe the inverters have rebooted?
[image]
I'm puzzled....

patience4711, 2023-04-06:
Most likely this is an coincident. The only way this can happen is that both ESP and inverter resetted. When you installed the software, the esp resetted. So indeed it looks like the inverters did so too. Maybe due to a grid failure? Or was this at the end of the day?

Anyway i trust that it works oke now?

Superbert25, 2023-04-06:
Time will tell. 

It was indeed at the end of the day…but there was still sufficient light for a bit of power production.

I’ll keep monitoring and report back here

patience4711, 2023-04-06:
@Superbert25   You seem to use the cc2530/2591 zigbee module, can you tell which firmware you flashed on it?

swbouman, 2023-04-06:
@patience4711 I’m also using the cc2530+2591 setup. I flashed the 2530+2591 firmware, available by koenk. 

I had to convert the .hex file to a flashable bin file. 
the 2530+2591 setup is running on standard 2530 firmware, but the amplifier is. It activated at that moment. 

Right now I achieved a range of 20 meters and the device is mounted in house, below the ceiling. 

Although I have to aim the antenna better hence the roof is reflecting the signals when the sun is on a certain angle.

closed 2023-04-11 by patience4711