patience4711/read-APSystems-YC600-QS1-DS3 #101: WIFI does not connect

pixeldoc2000, 2023-04-09:
Can't get WIFI to connect anymore.

My ECU was running fine for about a year+.

Latest Update from 9.6 to 9.6c failed. I've had to reflash my ESP8266. After reflash I can't get it to connect to my Wifi / WLAN anymore, no matter what.

What I have tried already:
- Flashed different ESP8266 -> Same Problem
   - Same two ESP8266 connects to my WIFI without changed Settings without any issue with ESPHOME Firmware
- Change WIFI Settings
   - Disabled 5G
   - Changed to WPA from WPA2/3
   - Changed Password und SSID
- Tried several Firmware Version from v9.8 to v9.3

Looking info compiling firmware myself with more serial debug messages enabled to get a clue why it is not connecting.

Any suggestions what else to try?

kentbrock, 2023-04-10:
Hi,

have you tried a clean up with the Espressif Tool?

https://www.espressif.com/en/support/download/other-tools

Sometimes I face this issue in other projects. A factory wipe always helps.

BR

pixeldoc2000, 2023-04-10:
@kentbrock 

Do you mean "Flash Download Tools" by Espressif? I don't see a way to do a factory reset.

But I already did a couple of erase_flash with esptool: `esptool --chip esp8266 erase_flash` which should do (almost) the same?

joffito, 2023-04-11:
try a power cycle and after powering on press the flash button. hold it till the led lights on and off (might be several times). the  release flash button and power cycle.

patience4711, 2023-04-11:
I am going to close this, it is a duplication of https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/66

closed 2023-04-11 by patience4711

pixeldoc2000, 2023-04-11:
@patience4711 I have seen that one, but the Solution to erase the flash did not work for me. That's why I opened a new issue.

@joffito did that the power cycle and try to enter flash mode connecting D3 / GIO0 and GND (d1 mini has no flash button) but the led light did not come on. I had to reflash the wemos d1 again. Did not make a differents.

Serial Log:
```
[12:20:26]/wifiCon requested
[12:20:26]send confirm page  
[12:20:32]could not connect, try again
[12:23:33]lijst = 
[12:23:33] 
[12:23:33]  <div><a href="#p" onclick="c(this)">FOO</a>&nbsp;<span class="q ">-61dBm</span></div>
[12:23:33]   
[12:23:33]  <div><a href="#p" onclick="c(this)">BAA</a>&nbsp;<span class="q ">-86dBm</span></div>
[12:23:33]  
[12:23:43]/wifiCon requested
[12:23:43]send confirm page  
[12:23:50]could not connect, try again
```
This will continue infinitely.

Look like the WIFI credentials are not saved. After reset, it will always enter "Portal Loop".

joffito, 2023-04-11:
just get yourself a 5 $ NodeMCU 8266 and enjoy ;-)

pixeldoc2000, 2023-04-12:
@joffito I tested a nodemcu, same issue, so no joy for me.

Simple Test Webserver with Arduino Example works also on my wemos d1 mini and nodemcu (like ESPHOME), so no Hardware Error.

So it must be a problem with the ECU Firmware.

swbouman, 2023-04-12:
It’s too bad that you experience these errors. Did you first erase whole esp? Sometimes this gives an error. 

What version did you flash? Did you try more versions?

pixeldoc2000, 2023-04-12:
@swbouman as I wrote before, yes erased the flash multiple time on multiple Devices, tried multiple ESP (they all work with every other Firmware but ECU), tried multiple Versions of ECU Firmware.