patience4711/read-APSystems-YC600-QS1-DS3 #110: Strange value in Domoticz

LordDarkneo, 2023-04-23:
Hi there,
As evrything's working quite fine now on my installation, I just realized that sometimes, I have "wrong" values in my Domoticz sensor. It happens many times that I have a "negative" consumption. Example yesterday:

[image]

Any idea on how to check why this value is stored / sent by the ESP?

Thanks for your help

joffito, 2023-04-23:
This is domoticz behaviour, not the ESP ECU. You need to create a sensor Electric (Current + Counter) and use the values from the screenshot
[image]

LordDarkneo, 2023-04-24:
Hi and thanks for your reply. 
I though it might be linked to a wrong data sent by the ESP... The only difference is the read value I set to "Computed".
[image]
I do not really understand what's the difference between the 2 options (even reading the documentation).... Let's see if the behavior is still then.

closed 2023-04-24 by LordDarkneo

joffito, 2023-04-24:
someone explained that behaviour in another issue. i think when you set to 'from device' then domotic sees a 0 value on ecu reset and then subtracts your current generation from 0 (as to calculate the difference between readings) leading to a negative value.