patience4711/read-APSystems-YC600-QS1-DS3 #111: Alternative firmware for ECU-B?

narc-Ontakac2, 2023-05-06:
This might (I am not good with hardware) be used to create an alternative firmware (no cloud, no mobile app, no registration) for the ECU-B. If this is doable I would very much appreciate it. I own one and can look into it if that helps.

narc-Ontakac2, 2023-05-09:
I already have an ECU-B and a DS3. When I started to set it up, which is a quite annoying process, I found out that it will give me power readings every 5 minutes. That is as useful to me as once a day, so I stopped and started looking for alternatives. One is yours. I can send a photo and would appreciate a comment on the hardware.

The actual problem I am trying to solve is that I get data from my power meter every 2 seconds. This was quite interesting but is now totally clobbered by the power produced by my DS3. To subtract that I need a resolution that is better than a minute.  The likely solution is an smart plug with tasmoga.

patience4711, 2023-05-09:
@narc-Ontakac2 This software has a data resolution of 5 minutes, a higher resolution causes stability troubles so i can't help you with this.

narc-Ontakac2, 2023-05-09:
Thanks, that is really useful information.

closed 2023-05-10 by patience4711