patience4711/read-APSystems-YC600-QS1-DS3 #112: NEW VERSION 9_12

patience4711, 2023-05-10:
In this new version the "engine" has been redesigned, resulting in significant (~40%) more free heap. Some small (cosmetic) issues are solved. All the functionallity has been tested and it works fine. 

**If you are currently using an older version:** 
 - Before updating put off the auto polling. 
 - After updating re-pair your inverters or edit the id via the console: eg: if inverter 0 has the id = 0xA1B2 you can change this with 10;edit=0-B2A1.  0xA1B2 should become B2A1.
This is a bit inconvenient but the change in the notation of the inv ID has many advantages as to free heap and program space.  

At this moment i don't think than i can do anything else to improve the heap so for now this is hopefully the last revision.
[v9_12](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_12.bin) 

[image]

swbouman, 2023-05-10:
@patience4711 Thank you for the new version. I renamed all inverters as described, everything works except the mqtt every inverter is send to 1 topic. Did you change anything in the mqtt?

patience4711, 2023-05-10:
@frtz13  Like I said, some values can get corrupt due to the change in the filesystem. So it is important to examen your settings.

frtz13, 2023-05-10:
so far everything looks good. heap and stack free memory looks stable, too.

>> after checking the "auto polling" checkbox again, I had to reboot the ESP-ECU to make auto polling happen again.

> I cannot reproduce that, maybe you didn't save the settings.

it must have been saved, otherwise it would not have started after the reboot. however, this happened after all the configuration changes, manual poll checks etc.

frtz13, 2023-05-11:
@patience4711 9_12 looks good after 24h. resetCounter=1. free stack stable at 104 B. free heap gradually came down from 26000 to slightly over 24000 B.

hvdb, 2023-05-11:
@patience4711  So I installed it all just yesterday (completely new)

And it worked, but now I get errorcode 11.
`

poll inverter 0
--
polling
zb send poll cmd inverter 0
sendSB sent FE1D24016CFC1414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFEB9
readZB FE0164010064FE034480E914013B rc=14
no AF_DATA_CONFIRM
polling failed 11

`
When I re-pair it works for a moment, but after that it fails again.
Any idea why?

inverter: DS3

Thanks for all the work!

patience4711, 2023-05-11:
@hvdb  What version did you install? What hardware? Are your settings oke? ECU too far away from the inverter?

patience4711, 2023-05-11:
@frtz13  Yes i had a zigbee reset too, due to a typo the timing of the healthcheck relative to the polling. I doesn't bother much so i'l  correct this in the next update. The heap in my test is fairly stable, going slightly down due to the increase in log messages. They are limited so that in longer term it should stay stable.

hvdb, 2023-05-11:
> @hvdb What version did you install? What hardware? Are your settings oke? ECU too far away from the inverter? Before polling first issue 10;init_n
> 
> Something to try: downgrade to [v9_10](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_10.bin) if you do; Put off autopolling first and re-pair after the software udate.

- Version 9_12
- nodemcu & cc2530
- Settings seems fine, it does work for a while
- Fairly close, signal strength around 10/12% I see now 

After the `10;init_n` it works again, let's see for how long.
At least saves me from re-pair :)

Will let you know!

hvdb, 2023-05-11:
@patience4711 Alright, updated.
Will keep an eye on it and move it a bit closer with less obstacles to improve the signal.

patience4711, 2023-05-12:
Version [v9_12b](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_12b.bin) solved the zigbee restarts at nighttime transition.

closed 2023-05-15 by patience4711