patience4711/read-APSystems-YC600-QS1-DS3 #114: NEW VERSION 9_13

patience4711, 2023-05-15:
The latest version now is  [v9_13](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_13.bin)
The last revision is a security improvement, we cannot access the configuration pages from outside our own network. So we can open it to the world with more peace of mind.

I have it running for some days now and it seems rockstable. The log shows no zigbee resets. When there are no webclients, the heap is steady over 26000. No memory leakage, 

Via [this link](http://hansiart.mooo.com:1222/) you can visit my testenvironment to see how it works. You can login as "user" and "2222"

swbouman, 2023-05-15:
Did you just reset your device? Or did it crash? 

I have difficulties connecting the webserver since 9.12 version. 

[image]

patience4711, 2023-05-15:
I uploaded a changed software. I already said before, there is something with the mosquitto. I also saw some disconnections this morning. Which is strange after 2 days of working problemless.  I'l investigate..

patience4711, 2023-05-15:
@swbouman  I meant that I installed something on my esp that caused it to reboot. 
Anyway i cannot see what is wrong with the mqtt, its all implemented as it should.

When the mqtt connection fails, it tries to reconnect once in 5 seconds, this shouldn't bother the wifi but I changed that period to 10 sec's.  I Also updated the pubsub library and changed the keepalive value (from 15 to 150 secs). I cannot think of anything else i could try. It is running now, must wait for results. 

If you want to test yourself [here it is](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_13_MQTT.bin)

swbouman, 2023-05-16:
I think it has something to do with the zigbee. It's impossible to pair the inverters anymore. The program changes the inverter type to YC600 and the inverter ID is always 1111. The program is not interupted and hangs on the pairing progress..

sendSB sent FE1824020FFFFFFFFFFFFFFFFF14FFFF140C0201000F060070400030337
--
analyzing paircmd
readZB nothing received
no pairing answer
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100704000303A3D810FFFF80971B01A3D8
sending paircmd 2
sendSB sent FE2324020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100704000303A3D810FFFF80971B01A3D879
analyzing paircmd
readZB nothing received
no pairing answer
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D8
sending paircmd 3
sendSB sent FE1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
success, inverter got id 1111
sending N.O. cmd
sendSB sent FE162401FFFF1414060001000F1E80971B01A3D8FBFB1100000D4F
readZB FE0164010064FE034480001401D2 rc=14
sendSB sent FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011 rc=19
found 0709, running oke

patience4711, 2023-05-16:
@swbouman What version are we talking about now? You get no answers on the pairing requests. This is probably the old range problem. The 1111 is an indication of failure, not success, so there is something wrong indeed. But if you would get answers, the pairing works oke. Why are you pairing, normally no need for? If you know the id you can edit this via the console. 

I have tested the pairing a thousand times and it worked better than ever. The thing is that i cannot reproduce the error because I **always** get answers on the pairing. But i found the bug, there was a return(50) on a bool function. I corrected that but this only comes into the play if there are no pairing answers. So the real problem is still the lack of answers.

[here is 9_13a] (https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_13a.bin)  with the correction.

swbouman, 2023-05-16:
@patience4711 I discovered a problem with 9.13 and the pairing. I needed to refresh the esp due to getting unresponsive behavior. I did get an answer but it was allways the first 4 chars of the ECU ID what was written as correct inverter ID. After editing the correct inverter id, which was displayed in the console output during the pairing, the inverter type changed to YC600 instead of the correct DS3 type. It took me a long time to get them all in line again. 

After a day of functioning the ESP-ECU webinterface is unresponsive again :( I saw a free heap of 13000 which should be high enough but still a lot lower than the 27000 at reboot. 

So two bugs: 

- getting unresponsive after hours of uptime
- changing the inverter type to YC600 after an EDIT=0-.... command.

Ill try 9.13b

joffito, 2023-05-16:
yeah i can confirm the switch to YC600. but i like the updated webinterface !
still stable after 5 h ontime.

patience4711, 2023-05-16:
@swbouman  indeed edit did the same, writing a too long string. This is also solved now in version [9_13c]{https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_13c.bin)

swbouman, 2023-05-16:
Thanks! @patience4711 !

The unresponsive behavior is probably due to a too busy fixed wifi channel. I changed the channel and the esp is responsive again. 

I'll test the EDIT string on version 9.13c

swbouman, 2023-05-16:
@patience4711 with ver 9.13 this is the output (for as far as I could reproduce it from the console)

`

found invID in MessageToDecode
--
found invID = A1BA
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100704000161600A3D810FFFF80971B01A3D8
sending paircmd 2
sendSB sent FE2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100704000161600A3D810FFFF80971B01A3D833
analyzing paircmd
readZB FE0164020067FE25448100000F010000141401240014B600000011704000161600A3D810FFFF80971B01A3D87C4B0D6F rc=48
decoding : FE0164020067FE25448100000F010000141401240014B600000011704000161600A3D810FFFF80971B01A3D87C4B0D6F
found invID in MessageToDecode
found invID = A3D8
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D8
sending paircmd 3
sendSB sent FE1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
success, inverter got id A3D8
sending N.O. cmd
sendSB sent FE162401FFFF1414060001000F1E80971B01A3D8FBFB1100000D4F
readZB FE0164010064FE034480001401D2 rc=14
sendSB sent FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011 rc=19
found 0709, running oke

`

the found inverter ID differs from time to time and as you can see, the written ID is the same as last 4 digitsof the ECU ID.

patience4711, 2023-05-16:
in the decoded string, the quoted part is your serial nr?  Than the id should be A3D8 (2 bytes behind the serial. Is that coincidence
 FE0164020067FE25448100000 F010000141401 240014B600000011 "70 40 00 16 16  00" A3D810FFFF80971B01A3D87C4B0D6F

when i pair a fake  inverter with serial 111111111111 i get this:
FE0164020067FE25448100000F010000141401 780072A900000011 11 11 11 11 11 11 B3D610FFFF80971B01B3D63A100D67
which evalutes to id B3D6 which is part of my ecu id

When i pair my real inverter I get a much longer answer that contains the serialnr twice, behind the first occurence the B3D6 belongs to my ecu, behind the second one the sought id.
FE0164020067FE25448100000F0100001414017B00F7A900000011 "408000158215" B3D610FFFF80971B01B3D63A100DA3FE1C4481000001013A101414007B0054AD00000008FFFF "408000158215" 3A100E1F rc=81 

So the answer isn't correct. The strange thing is that it works this way as long as the project exists. I can build an evaluation that rejects the message when too short but i must study that first.

swbouman, 2023-05-16:
My Ecu Id is D8A3011B9780  and the serial of the paired inverter is 704000158937 the working id is A1BA but was 9BDE earlier with the same Ecu id.

patience4711, 2023-05-16:
paircommand 1 and 2 should contain that serial but it doesn't. 

snprintf(pairCmd[2], sizeof(pairCmd[2]), "24020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100%s%s10FFFF%s", Inv_Prop[which].invSerial, ecu_short, ecu_id_reverse);

sendSB sent FE2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100 **704000161600** A3D810FFFF80971B01A3D833

So it looks as if Inv_Prop[which].invSerial contains 704000161600

swbouman, 2023-05-16:
Well, I do have that serial in the inverters. But it was not the inverter that was pairing. For the found serial I have inv Id 9BDE (704000161600) is everything messed up in my config?

patience4711, 2023-05-16:
I don't know what happened but maybe your filesystem is corrupted. Or you where pairing another inverter than you thought. 

Meanwhile i implemented something to prevent that an inverter gets an id equal to the ECU short id. Tomorow I'l test it.

patience4711, 2023-05-17:
@swbouman  I changed it in such way that pairing answers that are not long enough are not analyzed. I can't pair any non-existant inverters anymore. My own inverter paired without problems in one shot. 

[v9_13d](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_13d.bin)

swbouman, 2023-05-17:
I give it a shot! Thanks for the effort!

swbouman, 2023-05-17:
this is my output of polling inverter with id: 704000159303

readZB FE0164020067FE1A448100000C0200001414015000F3A1000000067040001593039FDD0D2E rc=37
--
decoding : FE0164020067FE1A448100000C0200001414015000F3A1000000067040001593039FDD0D2E
no valid pairing code, returning...
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100704000159303A3D810FFFF80971B01A3D8
sending paircmd 2
sendSB sent FE2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100704000159303A3D810FFFF80971B01A3D8B6
analyzing paircmd
readZB FE0164020067FE25448100000F0100001414014E0051BC00000011704000159303A3D810FFFF80971B01A3D89FDD0DBAFE1C448100000101534C141400120074C300000008FFFF704000159303534C0ECF rc=81
decoding : FE0164020067FE25448100000F0100001414014E0051BC00000011704000159303A3D810FFFF80971B01A3D89FDD0DBAFE1C448100000101534C141400120074C300000008FFFF704000159303534C0ECF
found invID = 534C
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D8
sending paircmd 3
sendSB sent FE1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
success, inverter got id 534C
sending N.O. cmd
sendSB sent FE162401FFFF1414060001000F1E80971B01A3D8FBFB1100000D4F
readZB FE0164010064FE034480001401D2 rc=14
sendSB sent FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011 rc=19
found 0709, running oke

swbouman, 2023-05-17:
and this for the inverter with serial: 704000159286

readZB FE0164020067FE1A448100000C0200001414018F00738F00000006704000159286A1BA0D82 rc=37
--
decoding : FE0164020067FE1A448100000C0200001414018F00738F00000006704000159286A1BA0D82
no valid pairing code, returning...
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100704000159286A3D810FFFF80971B01A3D8
sending paircmd 2
sendSB sent FE2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100704000159286A3D810FFFF80971B01A3D832
analyzing paircmd
readZB FE0164020067FE25448100000F0100001414015000D4A900000011704000159286A3D810FFFF80971B01A3D82ED00D0CFE1C4481000001019FDD1414005000B6B200000008FFFF7040001592869FDD0EBA rc=81
decoding : FE0164020067FE25448100000F0100001414015000D4A900000011704000159286A3D810FFFF80971B01A3D82ED00D0CFE1C4481000001019FDD1414005000B6B200000008FFFF7040001592869FDD0EBA
found invID = 9FDD
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D8
sending paircmd 3
sendSB sent FE1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
success, inverter got id 9FDD

patience4711, 2023-05-17:
@swbouman  blow the horn!

swbouman, 2023-05-17:
🥳

patience4711, 2023-05-22:
No news is good news so they say....

closed 2023-05-22 by patience4711