patience4711/read-APSystems-YC600-QS1-DS3 #115: WiFi never connect

jayjay13011, 2023-05-16:
Hi!
your solution seems so good but I’ve got a big problem from the beginning I decided to install it. It’s the first time I can’t connect my Wemos D1 mini to my WiFi. I tried so much cards, new ones, with EspFlasher, on Linux, Windows, Mac and I can’t understand. I never had this kind of problem with others bin than this. And for information , I can flash with Tasmota…
Serial monitor say could not connect. Try again…
Help!

swbouman, 2023-05-16:
Hi @jayjay13011,

I don't want to sound patronizing, but are you trying to join a 2.4 GHz network? Wemos cards do not support 5 GHz.

I have flashed a Wemos multiple times from scratch using the provided .bin files. I never had an issue until last week. It turned out there was a disturbance on ch1 on the 2.4 GHz frequency at my home. That was the culprit, preventing any connection to the network.

The espflasher.exe also has a serial console, which displays everything thanks to the extensive built-in feedback in the program.

So please double-check if the network is reachable for the Wemos, make sure it is a 2.4 GHz network, and verify that the entered password is correct for the WiFi (check caps lock).

patience4711, 2023-05-16:
@jayjay13011 If it can't connect, an accesspoint is opened, the blue led is on. Please read the wiki

jayjay13011, 2023-05-16:
Yes when blue led is on, I can see the network ECU XXX, but when I try to connect it with my Wifi on the portal ;entering my Good password and selecting my 2.4Ghz network, the Wemos goes back in AP mode with message on serial communicator « could not connect. Try again ». Otherwise, you mention password 1234567890 for the AP of ESP but it’s an opened network. 
I’m pretty sure there is something bad in what I do, but I’m usually flashing cards and really, I can’t understand !
By the way, sorry for my poor FR-English!

patience4711, 2023-05-16:
You are not the first and likely not the last.  [read here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/93)

jayjay13011, 2023-05-17:
I tried all of these solutions. I tried again tonight but no success. One time, few days ago, I improved to go on the main config page. But I clicked on a link, Wemos disconnected and I never could reconnect. With no one of 6 cards I tried. Esptool, espflasher, arduino ide…

patience4711, 2023-05-18:
So i wonder what are you doing wrong when so many people can do this without problems.

When you could get on the main page, the flashing was successful so that is not the problem. Your wifi credentials are not saved however, so there must be an old filesystem on the ESP. Especially when you first had tasmota on it. You should completely erase the flash first! ! ! ! !
What file are you flashing exactly?
Did you try with a factory new one? 
Did you try this:` first flash another program, (with Flash Size configuration 4MB(FS:2MB OTA:~1019KB)) for example a basic web server, with the wifi credentials hardcoded. If the esp connects than flash the ESP-ECU.bin without erase memory.`
You have a normal esp8266 nodeMcu (not a esp8285 or esp32) ?
What is the IP of your router 192.168.x.x ? Not some odd configuration?
If you see the blue led, try to connect to another network, p.e. in your friend's house. To rule out that your network is the cause.
Try to flash [this bin](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-CUCKOO-v1_0.bin) to see if that works.
What do you use the nodeMcu's for normally?

barbu34, 2023-05-18:
Hi all,
I have got the same problem. ESP8266 arduino D1 R2 with CC2530 and DS3-L, and a wifi box with 802.11g and 802.11ac 5ghz and 2.4Ghz 
I flashed the ESP with ESP-ECU_v9_13d.bin. I flashed CC2530 with CC2530ZNP-with-SBL.bin
I had reset the ESP and i found the ECU-XXXXXX on my computer (Window11). I selected connection and without password the connection passed to connected but limited.
When i run http://192.168.4.1 the board ask me to enter the SSID and the password. I enter the name of my wifi network and the WPA2 password. The page change and ask me " CONNECT TO WIFI please wait until the led has flashed, then click OK". I wait 2 or 3 min and after the led flash i check the comunication with my internet box com is ok but the ECU is no more in my wifi list on my computer, and i click OK and after 30s the web page say delay is over try an other time. When i try no more change.
The serial arduino console say 
-> Failed to open for read/Inv_Prop0.str
-> booted up, checking coordinator
-> 192.168.x.xx
-> swapping serial to zb, goodbye

It is strange because when i open the OS box tools i see the ECU connected with WPA2 authentification continuing some transmitions Rx of few octets but no Tx. And the ECU disappeared from the list of wifi. And even after a hard reboot of the ESP the wifi connection don't come back in the wifi list and the web board refuse to start. To try an other time i need to reset the flash with the erase flash setting to "sketch+wifi setting" and reflash after. 
It is possible that the CC2530 flash is not good ?
or the connecting wires are not good ? (CC2530 pin RST to arduino D5, CC2530 PO2 to arduino D8, CC2530 PO3 to arduino D7, VCC to 3.3v, GND to GND)
Some ideas for what i can try.
I try the last flash bin just over but same problem.

patience4711, 2023-05-18:
@barbu34 if you see an IP 192.168.etc in the boot info, then it is connected to WIFI.  Just browse to that ip.  Open your browser and type "http://192.168...etc"
This has nothing to do with the cc2530, you can do the wificonnection without the cc2530 wired.

barbu34, 2023-05-18:
When i enter this IP on the broswer it ask me a login and a password. So witch login and witch password ? Master with 0000 ?

fwolfst, 2023-05-18:
> When i enter this IP on the broswer it ask me a login and a password. So witch login and witch password ? Master with 0000 ?

Yes, admin and 0000 if you did not change that in the screen before.

frtz13, 2023-05-18:
@patience4711 flashed v. 9_13d to a brand new ESP. from the initial Wifi config. page, after clicking ok, the console always stated: "could not connect, try again". the web page returned me to the Wifi config page. Did this 3 times. Then I rebooted disconnecting power. The ESP came up with its IP address, has a Wifi connection, and seems to work fine ever since (without Zigbee, though).

fwolfst, 2023-05-18:
> @patience4711 flashed v. 9_13d to a brand new ESP. from the initial Wifi config. page, after clicking ok, the console always stated: "could not connect, try again". the web page returned me to the [...]

I think the esp8266s wifi is a bit unstable, it does not always connect for me on the first try either.

barbu34, 2023-05-18:
So thanks, with admin for login and 0000 for password it works.
Now i will try to follow and understand the readme to continue and connect all my DS3-L

jayjay13011, 2023-05-18:
> So i wonder what are you doing wrong when so many people can do this without problems.
>
> When you could get on the main page, the flashing was successful so that is not the problem. Your wifi [...]

Here is the result of serial with a basic web server with hardcoded credentials:
```
19:37:01.732 -> Connecting ...
19:37:09.149 -> ..
19:37:16.609 -> 
19:37:16.609 -> Connected to Freebox-67F1A8 2.4GHZ
19:37:16.609 -> IP address:	192.168.1.36
19:37:16.609 -> mDNS responder started
19:37:16.609 -> HTTP server started
```

For you information, the program I used - You can see my credentials (those I use for your soft...)
```ruby
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WiFiMulti.h> 
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>

ESP8266WiFiMulti wifiMulti;     // Create an instance of the ESP8266WiFiMulti class, called 'wifiMulti'

ESP8266WebServer server(80);    // Create a webserver object that listens for HTTP request on port 80

void handleRoot();              // function prototypes for HTTP handlers
void handleLogin();
void handleNotFound();

void setup(void){
  Serial.begin(115200);         // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');

  wifiMulti.addAP("Freebox-67F1A8 2.4GHZ", "WpDggYjy6rshgHPu5i");   // add Wi-Fi networks you want to connect to

  Serial.println("Connecting ...");
  int i = 0;
  while (wifiMulti.run() != WL_CONNECTED) { // Wait for the Wi-Fi to connect: scan for Wi-Fi networks, and connect to the strongest of the networks above
    delay(250);
    Serial.print('.');
  }
  Serial.println('\n');
  Serial.print("Connected to ");
  Serial.println(WiFi.SSID());               // Tell us what network we're connected to
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());            // Send the IP address of the ESP8266 to the computer

  if (MDNS.begin("esp8266")) {              // Start the mDNS responder for esp8266.local
    Serial.println("mDNS responder started");
  } else {
    Serial.println("Error setting up MDNS responder!");
  }

  server.on("/", HTTP_GET, handleRoot);        // Call the 'handleRoot' function when a client requests URI "/"
  server.on("/login", HTTP_POST, handleLogin); // Call the 'handleLogin' function when a POST request is made to URI "/login"
  server.onNotFound(handleNotFound);           // When a client requests an unknown URI (i.e. something other than "/"), call function "handleNotFound"

  server.begin();                            // Actually start the server
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();                     // Listen for HTTP requests from clients
}

void handleRoot() {                          // When URI / is requested, send a web page with a button to toggle the LED
  server.send(200, "text/html", "<form action=\"/login\" method=\"POST\"><input type=\"text\" name=\"username\" placeholder=\"Username\"></br><input type=\"password\" name=\"password\" placeholder=\"Password\"></br><input type=\"submit\" value=\"Login\"></form><p>Try 'John Doe' and 'password123' ...</p>");
}

void handleLogin() {                         // If a POST request is made to URI /login
  if( ! server.hasArg("username") || ! server.hasArg("password") 
      || server.arg("username") == NULL || server.arg("password") == NULL) { // If the POST request doesn't have username and password data
    server.send(400, "text/plain", "400: Invalid Request");         // The request is invalid, so send HTTP status 400
    return;
  }
  if(server.arg("username") == "John Doe" && server.arg("password") == "password123") { // If both the username and the password are correct
    server.send(200, "text/html", "<h1>Welcome, " + server.arg("username") + "!</h1><p>Login successful</p>");
  } else {                                                                              // Username and password don't match
    server.send(401, "text/plain", "401: Unauthorized");
  }
}

void handleNotFound(){
  server.send(404, "text/plain", "404: Not found"); // Send HTTP status 404 (Not Found) when there's no handler for the URI in the request
}
```

And here is the result when I try with your bin just after
```
~ % % python3 -m esptool --port /dev/cu.usbserial-1420 --chip esp8266  write_flash -fm dio 0x00000 /Users/jeremylopez/ESP-ECU_v9_13d.bin
fg: no current job
jeremylopez@iMac ~ % python3 -m esptool --port /dev/cu.usbserial-1420 --chip esp8266  write_flash -fm dio 0x00000 /Users/jeremylopez/ESP-ECU_v9_13d.bin
esptool.py v4.5.1
Serial port /dev/cu.usbserial-1420
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:e7:29:49:40:c4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00078fff...
Compressed 495536 bytes to 327049...
Wrote 495536 bytes (327049 compressed) at 0x00000000 in 31.8 seconds (effective 124.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
And the serial when I enter my credentials on the web page
```
19:48:50.271 -> entering portal loop
19:49:45.417 -> lijst = 
19:49:45.417 ->  
19:49:45.417 ->   <div><a href="#p" onclick="c(this)">DIRECT-97WorkCentre 3225</a>&nbsp;<span class="q ">-47dBm</span></div>
19:49:45.450 ->    
19:49:45.450 ->   <div><a href="#p" onclick="c(this)">Freebox-67F1A8 2.4GHZ</a>&nbsp;<span class="q ">-63dBm</span></div>
19:49:45.450 ->   
19:50:04.277 -> /wifiCon requested
19:50:04.341 -> send confirm page  
19:50:10.514 -> could not connect, try again
```

I hope it helps you see clearly my problem!!!!

Thanks for all

jayjay13011, 2023-05-18:
New element:
I just tried with direct wifi of my printer and it works! (yoepy, connected)
So the flash is ok, the problem is with my network.
But what can be the problem with my network I have with your bin, and I don't have with Tasmota or others???
And, unfortunately I can't stay on my printer wifi for this card.

patience4711, 2023-05-19:
If you installed the basic webserver with Flash Size configuration 4MB(FS:2MB OTA:~1019KB)
and connected with that: 
Now flash the ESP-ECU bin **without** erasing flash. Now it can connect with the already present credentials.

Do you have spaces in the network ssid? Maybe that is the problem. Is that name shown in the scanned networks. And did you select it from that list? Try to type in the SSID yourself or you could try to put the name in singlequotes like **'Freebox-67F1A8 2.4GHZ'**  but i don't know if that helps. Or change the SSID to Freebox-67F1A8_2.4GHZ or setup a guest network with an SSID without spaces.
What else could be wrong with your network? Maybe try another wifi channel? It could be some other setting in your router? 
If you go to the neighbours, relatives, friends, and try to connect on their network and it works, maybe you can compare their settings to see where it differs.

frtz13, 2023-05-19:
@patience4711 there seems to be something misleading in the initial wifi connection procedure. Here is what I did:
flash v9_13 wiping all flash on a D1 mini
power cycle ESP and set up serial console (115200, 7bit)
power cycle again to get console messages at boot
on console: scan start, list of wifi networks, entering portal loop
on phone: connect to ESP Access point, then to 192.168.4.1 to go to HANSIART WIFI CONFIG page
on phone: click on WIFI CONFIGURATION
on phone: shows list of wifi networks.
on console: html text with list of wifi networks (lijst = … )
on phone: fill out the form (all 3 fields manually). no spaces in wifi network or passwords. click save.
on ESP:  LED switches off, comes back on after approx. 1 second
on phone: CONNECT TO WIFI page. (please wait until…)
on console: could not connect, try again
on my wifi router: ECU is connected and got its IP address on my wifi network
Note: if you click OK now on the phone, the next page will say that you are _not_ connected to a wifi network, and will let you go back to the WIFI CONNECTION page.
power cycle ESP
on console:
leesStruct ivn = 0
Failed to open for read/Inv_Prop0.str
booted up, checking coordinator
192.168.x.y
swapping serial to zb, goodbye

now the ESP is up and running normally. however, we never get this confirmed in the initial wifi configuration procedure visible on the phone, not even in the serial console (before power cycling).
to write up this report, I went through this half a dozen times. it always happened in exact the same steps.

patience4711, 2023-05-19:
@frtz13  Thats because there is a short disconnection when changing from softAP to STA to connect and than go back to softAP to show us the IP. 

A windows computer reconnects in that case to the softAP but a phone doesn't. Unless you tell it to automatically always connect to 192.168.4.1 when available. 
So because of that you miss the confirmation page.

This is very complicated stuff this is why i described the trick with the resetbutton when the connected message is shown in the serial monitor. 
I added a line in the webpage "please wait ... etc  Or press the resetbutton on your ESP when nothing's happening. So that makes it less misleading. 

I am still waiting for a library that works 100% in all possible situations.

frtz13, 2023-05-19:
@patience4711 
> So what you say is that the connection fails but after a powercycle it connects? So without the serial message and the ledflashes?

... what I say is that:
- the console says that the connection fails.
- the wifi router shows that the connection has been established,
- the LED switches off,, then on again steadily. no three flashes.
- after power cycling the ESP connects and works.

patience4711, 2023-05-19:
Yes, that's misleading.  It survived 10 evaluations of **WiFi.status() != WL_CONNECTED** with a delay(500) between them. Any suggestions on what i could do about that? Maybe more attemps? Longer delays?
I don't know how WiFi.status() works but i have nothing else to rely on. 

Anyway i can't reproduce it. With a serial message between the attempts i always get 
```
send confirm page  

wifi state = 6
wifi state = 6
wifi state = 6
wifi state = 3yoepy, connected
```
Also on an ESP flashed with erase of all flash contents. It connects after 4 attempts.  

Tried with my phone even better, works exactly as it should
/wifiCon requested
send confirm page  

wifi state = 6
wifi state = 6
wifi state = 6
wifi state = 3
youpy, connected
portal timed out, resetting...

So i doubled the delays between the connection evaluations and added a debug message so that we beter know what's going on. Maybe not the solution in your case but who knows [v9_14exp](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_14exp.bin)

frtz13, 2023-05-19:
@patience4711 yes, this looks nice to me.
on the console, instead of the "could not connect, try again" message, I now get what you described (wifi state =.... youpy, connected).
on the ESP: I get the LED flashes now.
on the phone: after clicking OK, I get the confirmation page with the IP address.
thanks @patience4711 !
BTW: I always erased flash memory with NodeMCU_PyFlasher / esptool.py v.3, just before flashing it with the binary via the serial connection.

patience4711, 2023-05-19:
@frtz13  Great although i can't explain that. After how many attemps of what wifi state? Again i don't know how WiFi.state() works but it seems that it polls something at your router. Maybe needed more time to get an answer from your router. 

This version has some more efficieny in the way the data requests are processed, by the program and by the javascript in the webpage. It should have shorter reactiontimes.

frtz13, 2023-05-19:
@patience4711 this is the full console output:
wifi state = 6
wifi state = 6
wifi state = 6
wifi state = 6
wifi state = 6
wifi state = 3
youpy, connected

jayjay13011, 2023-05-19:
> @jayjay13011 The problem seems that your network is not accepting the credentials. Do you have spaces in the network ssid? Maybe that is the problem. Is that name shown in the scanned networks? And did you select it from that list? You can try one of the following:
> 
> * type in the SSID yourself insteat of selecting it.
> * type the name in singlequotes  (') like **'Freebox-67F1A8 2.4GHZ'**
> * change the SSID to Freebox-67F1A8_2.4GHZ
> * setup a guest network with an SSID without spaces.
> 
> What else could be wrong with your network? Maybe try another wifi channel? It could be some other setting in your router?
> 
> If you installed the basic webserver with Flash Size configuration 4MB(FS:2MB OTA:~1019KB) and connected with that: Now flash the ESP-ECU bin **without** erasing flash. Now it can connect with the already present credentials.

Ok, I typed the SSID, type in singlequotes, no change.
I can't change my SSID because I have about 50 devices in my network which are synchronized with my credentials. 
Also, my guest WiFi can only be in 5Ghz...
But my WiFi printer has space in his SSID. And I have no problem.
I just tried with the new bin and another new ESP. Here is the result:
```
19:58:23.029 ->   <div><a href="#p" lijst = 
19:58:23.029 ->  
19:58:23.029 ->   <div><a href="#p" onclick="c(this)">Freebox-67F1A8 2.4GHZ</a>&nbsp;<span class="q ">-52dBm</span></div>
19:58:23.029 ->    
19:58:23.029 ->   <div><a href="#p" onclick="c(this)">DIRECT-97WorkCentre 3225</a>&nbsp;<span class="q ">-68dBm</span></div>
19:58:23.029 ->    
19:58:23.029 ->   <div><a href="#p" onclick="c(this)">Freebox-478D26</a>&nbsp;<span class="q ">-90dBm</span></div>
19:58:23.029 ->    
19:58:23.029 ->   <div><a href="#p" onclick="c(this)">FreeWifi_secure</a>&nbsp;<span class="q ">-91dBm</span></div>
19:58:23.029 ->   
19:59:01.579 -> /wifiCon requested
19:59:01.579 -> send confirm page  
19:59:03.304 -> 
19:59:03.304 -> wifi state = 6
19:59:05.072 -> wifi state = 6
19:59:06.845 -> wifi state = 6
19:59:07.773 -> wifi state = 6
19:59:08.664 -> wifi state = 6
19:59:09.541 -> wifi state = 6
19:59:10.446 -> wifi state = 6
19:59:11.352 -> wifi state = 3
19:59:11.352 -> youpy, connected
19:59:17.736 -> portal timed out, resetting...
19:59:19.474 -> 
19:59:19.474 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
19:59:19.474 -> 
19:59:19.474 -> load 0x4010f000, len 3456, room 16 
19:59:19.474 -> tail 0
19:59:19.474 -> chksum 0x84
19:59:19.474 -> csum 0x84
19:59:19.474 -> va5432625
19:59:19.474 -> ~ld
19:59:25.768 -> scan start
19:59:27.962 -> 3 networks found
19:59:27.962 -> 1: Freebox-67F1A8 2.4GHZ (-51)*
19:59:27.962 -> 2: Freebox-478D26 (-90)*
19:59:29.096 -> 3: Freebox-397C7A (-90)*
19:59:29.096 -> 
19:59:29.096 -> entering portal loop
```
So, as you can see, there is a changement. The web page give my new IP on my network, but when ESP restart, it goes on AP mode again and again.

Thanks for your help!

frtz13, 2023-05-19:
@jayjay13011 I created a Wifi network named: Freebox-67F1A8 2.4GHZ
on a D1 Mini (4 MB flash), using NodeMCU PyFlasher: erased flash and flashed v9_14exp.
on the phone, went through the wifi configuration steps: click on Freebox... in the list, then enter passwords.
In the console, got several wifi state =6 , then a wifi state =3 youpi, connected.
LED flashed. confirmed the Wifi connection in the web pages and took note of the IP address.
the ESP connects to the Freebox... network, and does so after power cycling, too, and works fine.

jayjay13011, 2023-05-20:
Ok. I finally improved. I don't know how, I never did something different, but it worked.
I could sync my inverters with my ZB, that was so easy compared to all the rest!
And for your information, I have 5 inverters DS3-L and it works.
But I prefer waiting several days before definitely transfer all my inverters from ECU-C to the soft, one time I will be sure stability of ESP is ok.
I will post as soon as I will do it!

closed 2023-05-25 by patience4711