patience4711/read-APSystems-YC600-QS1-DS3 #117: NEW VERSION 9_14

patience4711, 2023-05-22:
The latest version now is [v9_14](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_14.bin)

Via [this link](http://hansiart.mooo.com:1222/) you can visit my testenvironment to see how it looks. You can login as "user" and "2222"

I apologise in advance for eventual bugs but i have limited test opportunities, if have only one inverter of type yc600. For the other types i depend on the results of others who want to share their testing results.

The last revision is an improvement in the efficiency of data aquisition by the webpage. This should result in a shorter reactiontime. Also some tuning is done in the timing of the wifi config portal. 

Please note: If you are currently using a version older than 9_11:

Before updating put off the auto polling.
After updating re-pair your inverters or edit the id via the console: eg: if inverter 0 has the id = 0xA1B2 you can change this with 10;edit=0-B2A1. E.g. 0xA1B2 should become B2A1

I have it running for some days now and it seems rockstable. As you can see yourself. The log shows no zigbee resets. When there are no webclients, the heap is steady over 26000. No memory leakage,

FranZga, 2023-05-23:
Hi Patience.
I have a qs1, installed version 9_14 yesterday, I don't know why, but the ota upgrade didn't work and I had to use usb and a flash tool. 
Apart from that, the web interface is now incredibly fast, kudos! 
During the day I have experienced three connection failed states: -2, but probably due to the weakness of the zegbee signal which is only 30%, but this does not affect the data fruition.
So far so good, thank you very much!

frtz13, 2023-05-24:
@patience4711 running fine here for one and a half days now. zigbeeReset = 0. free stack = 120, free heap > 24000B (MQTT).
2 x YC600.
thanks.

chrostek, 2023-05-24:
How does your new security check introduced in 9_13 work? i cannot access my system in my local lan - my network is 172.30.0.0/19 which is a regular private network: https://en.wikipedia.org/wiki/Private_network

chrostek, 2023-05-24:
The IP of the ESP is 172.30.2.211 - if you compare the first 9 characters you do not support bigger networks (e.g. 255.255.0.0) - but i already tried my PC with 172.30.2.212 and it also gave me an "ACCESS DENIED!"

patience4711, 2023-05-24:
@chrostek if your router would only issue ips with 172.30.0. (i think you can change that in the settings) it wouldn't be a problem.

 If i go down to 7 characters that would would work for you i guess.  You can try this one 
[v9_15](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_15.bin)

swbouman, 2023-05-24:
A shot in the wild: probably the problem is in the 9 char comparison. Don’t know exactly how it works, but is the program compares the first 9 digits of the op address, the up address of @chrostek could give a false postitive. Maybe less chars to be the same as network of origin could help?

swbouman, 2023-05-24:
@patience4711 nice! Haha same idea 😜

chrostek, 2023-05-25:
only comparing characters would never work for everyone ... if someone has a 10.0.0.0/8 network the ESP could be 10.0.0.1 and the pc could be 10.254.254.254 and they are still in the same network ...

patience4711, 2023-05-25:
It's hard to please everybody but I am doing my best. As a workaround I can make the security level user configurable from 0-9. This variable can only modified via the accesspoint. A 0 means no security and a 7 means that the first 7 characters must match.

[image]

chrostek, 2023-05-25:
thanks - after half an hour with some tricks i got access to the ecu menu again to flash the new version - i like the option to disable it completely, because i would like to access the ecu also via vpn. works again for me 👍

patience4711, 2023-05-25:
Since the last reboot, the heap is almost stable at 25000

[image]

FranZga, 2023-05-26:
Hi Patience, great job, but something has changed in the MQTT out messages and now the red node can't parse the payload message correctly.  
[image]

patience4711, 2023-05-26:
@FranZga  sorry, something went wrong with the conversion from string to char. The format3 used to be like this: 
 {"inv_serial":"408000158211","acv":68.2,"freq":50.0,"temp":18.0,"dcv":[36.8,37.0],"dcc":[4.3,3.0],"pwr":[nan,nan],"totalen":174.35}
Wait an hour or so for me to fix this.

patience4711, 2023-05-26:
@FranZga  It should be oke now, checked and got {"freq":50.0,"temp":37.1,"acv":225.8,"dcv":[33.2,33.8],"dcc":[7.8,7.5],"pwr":[241.7,237.9],"en":[936.69,913.41]}
To answer your question: If you want to disable the output of the inverter itself, no. That should be done with (yet unknown) zigbee commands. 

[v9-16a](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_16a.bin)

FranZga, 2023-05-26:
[image]

> @FranZga It should be oke now, checked and got {"freq":50.0,"temp":37.1,"acv":225.8,"dcv":[33.2,33.8],"dcc":[7.8,7.5],"pwr":[241.7,237.9],"en":[936.69,913.41]} I fixed some other small issues too, (mqtt retain flag and changed some uses of strings) To answer your question: If you want to disable the output from one channel on the inverter itself, no. That should be done with (yet unknown) zigbee commands.
> 
> [v9-16a](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_16a.bin)
@patience4711 I have try but again was not able to parse the payload message like before.
Thanks for answering my silly question 😄

patience4711, 2023-05-26:
@FranZga   No, the problem is 1 comma in the energy array. I'd seen that already and overwrote the file but you were too quick. If you dowload and install it again it will be fixed.

FranZga, 2023-05-26:
@patience4711 Thanks, works perfectly now!

hvdb, 2023-05-28:
Works great here also.
So restarts anymore :)

One thing I noticed that in the API response,'energy' is now called 'eN'

Is that a wanted/expected change?

patience4711, 2023-05-30:
@hvdb Glad to hear that it works, How many inverters and what type you have? 
The eN is a typo, i'll change it back to "energy".

hvdb, 2023-05-30:
@patience4711 
I have 2 inverters, DS3

patience4711, 2023-06-09:
I am closing this now, there is a new version.

closed 2023-06-09 by patience4711