patience4711/read-APSystems-YC600-QS1-DS3 #120: NEW VERSION 10_2

patience4711, 2023-06-09:
In the latest version all use of strings in any of the main processes have been banned. There also are some changes as to the mosquitto implementation. An option is added to set the client id and an extra format is added so we can use thingspeak. This is 'nice to have' but not usefull if we have multiple inverters.<br>
Another mqtt option is the state idx; If this is not null, a mqtt message containing free heap and stack is send at every healthcheck.
If the outTopic has a trailing slash, the topic will be "outTopic/heap".

There is a mosquitto test option in the console that sends a test message.

Another novelty is a zigbee loopback test in the healthcheck instead of the ping. If the loopback fails, no attempt is done to start the zigbee coordinator. 

After some weeks of testing i am convinced that it works stable.  For those who want to test it: [v10_2](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v10-2.bin)

patience4711, 2023-06-12:
@swbouman I have an idea yes, the corruption of the filesystem is expected. Sadly this cant be avoided. After your full new install, was the problem with the cc2530 gone? The program now tests if there is a loopback answer and if not, there is no working zigbee. 
I installed it on a brand new un-used nodemcu some moments ago. At first it kept rebooting. After flashing 9_12 and upgraded to 10._2 the filesystem was corrupted. I tped 10;erase in the console to wipe all inverterfiles. Than i checked the other settings and made them correct.  Now i can add any inverter, no problems. 
But i still have to investigate...

swbouman, 2023-06-12:
@patience4711 thank you for your fast response. To be clear: I first installed the new version by upgrading from 9.16. Then I had the loopback errors. Afterwards I installed 9.16 again. Then I wiped the esp and installed 10.2 freshly. I started by adding the inverters. 

After filling all required fields and save the settings of the first inverter, the esp rebooted. When it cam back online and I looked ad inverter 1 serialnumber, the program changed the given serialnumber to zeros. And the inverter type was changed from DS3 to YC600. 

I changed back to fresh install of 9.16 and the given inverter numbers are correct saved. 

So I can’t save the inverter serial and type on a fresh installed 10.2 installation. And when the esp was updated the zigbee module is down all the time. 🥹

swbouman, 2023-06-12:
I followed your guide and successfully reinstalled version 10.2, but unfortunately, the zigbee issue persists. However, after updating to version 10.3, it seems that the problem with zigbee has been resolved. 

I would appreciate it if the inverter overview with pairing codes could be reintroduced on the infopage. Currently, I'm relying on screenshots for each inverter, which is quite challenging.

juliend20, 2023-06-14:
@patience4711 Thank you so much to add that. Great job ❤️

supersjimmie, 2023-06-28:
I updated from 9.6-something to 10.6 without any issue (the inverter was still connected).

closed 2023-07-29 by patience4711