patience4711/read-APSystems-YC600-QS1-DS3 #127: Pairing fail with DS3L

neruyzo, 2023-08-02:
Hi,

I use a noname D1 Mini flash with `ESP-ECU-v10_8` and a CC2530 + 2591 flash with `CC2530ZNP_2591-with-SBL.hex`.
The solar panel setup is 5 DS3L and 10 solar panels, I tried pairing different inverters but the behaviors are the same.

Below are commands and corresponding logs.

<details>
  <summary><code>10;init_n</code></summary>

```text
coordinator init cmd 0
sendSB sent FE03260503010321
readZB FE064180010202020702C1FE0166050062 rc=17
coordinator init cmd 1
sendSB sent FE0141000040
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2 rc=11
coordinator init cmd 2
sendSB sent FE0A26050108FFFF80971B01A3D856
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062 rc=6
coordinator init cmd 3
sendSB sent FE032605870100A6
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062 rc=6
coordinator init cmd 4
sendSB sent FE0426058302D8A3DD
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062 rc=6
coordinator init cmd 5
sendSB sent FE062605840400000100A4
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062 rc=6
coordinator init cmd 6
sendSB sent FE0D240014050F0001010002000015000020
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062FE0164000065 rc=6
coordinator init cmd 7
sendSB sent FE00260026
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062FE0164000065FE00660066FE0145C0088CFE0145C0088CFE0145C0088C rc=23
sending N.O. cmd
sendSB sent FE162401FFFF1414060001000F1E80971B01A3D8FBFB1100000D4F
readZB FE016401C2A6FE0145C0088CFE0145C0088CFE0145C0088C rc=24
sendSB sent FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011 rc=19
zigbee running oke
ZB coordinator started
```

</details>

<details>
  <summary><code>10;ZBT=2710AABBCC</code></summary>

```text
going to send a teststring, len=17
sendSB sent FE032710AABBCCE9
readZB FE036710AABBCCA9 rc=8
```

</details>

For the pairing, the web page will stay in the pairing process and the status of inverter (another tab) will be `1111` till the reboot of the ESP.

<details>
  <summary>Pairing an inverter</summary>

```text
coordinator init cmd 0
sendSB sent FE03260503010321
readZB FE064180010202020702C1FE0166050062 rc=17
coordinator init cmd 1
sendSB sent FE0141000040
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2 rc=11
coordinator init cmd 2
sendSB sent FE0A26050108FFFF80971B01A3D856
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062 rc=6
coordinator init cmd 3
sendSB sent FE032605870100A6
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062 rc=6
coordinator init cmd 4
sendSB sent FE0426058302D8A3DD
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062 rc=6
coordinator init cmd 5
sendSB sent FE062605840400000100A4
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062 rc=6
coordinator init cmd 6
sendSB sent FE0D240014050F0001010002000015000020
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062FE0164000065 rc=6
coordinator init cmd 7
sendSB sent FE00260026
readZB FE064180010202020702C1FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062FE0164000065FE00660066FE0145C0088CFE0145C0088CFE0145C0088C rc=23
sendSB sent FE00270027
readZB FE0E670000FFFF80971B01A3D8000007080010FE0145C0088C rc=25
starting ZB coordinator failed
pairing failed, zb system down
```

</details>

And the system information, with highlight on `zigbeeUp | 0` and `current errorCode | 11`.

<details>
  <summary>SYSTEM INFORMATION</summary>

```text
firmware version | ESP-ECU-v10_8
time retrieved | yes
systemtime | 15:17 summertime
system uptime | 0 d 0 h 28 m
wifi signalstrength | -57dB
ESP CHIP ID nr: | 13179454
Free heap | 23488 bytes
mqtt status | not configured
zigbeeUp | 0
autopolling | 1
polled | 000000000
current errorCode | 11
ZB resetCounter | 0
```

</details>

I read that some old DS3-L may have problems and need firmware update, if one of mine look like it may need an update the other 4 look more new and were not listed.

<details>
  <summary>Serial number inverter</summary>

```text
703000319xxx
703000734xxx
703000723xxx
703000723xxx
703000734xxx
```

</details>

I don't know where is my mistake, the zigbee module look like working from the init and the ping, but I fail to pair any inverter.
It looks like this setup is not new from previous issues with people make it work.

Thanks for the help.

patience4711, 2023-08-02:
@neruyzo  It looks like the coordinator fails to start in a special mode that is needed to pair inverters.  Could you install an older version to see if the problem persist?
[v9_16](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_16b.bin)

neruyzo, 2023-08-02:
@patience4711 

It's more verbose with `paircmd` send, failure at `paircmd 3` with`inverter got id 0000` response and `paircmd 1`/`paircmd 2` with `no valid pairing code, returning...`.

<details>
<summary>log</summary>

```text
init zb coordinator
ZBmodule hard reset
coordinator init cmd 0
sendSB sent FE03260503010321
readZB FE0166050062 rc=6
coordinator init cmd 1
sendSB sent FE0141000040
readZB FE0166050062FE064180020202020702C2 rc=11
coordinator init cmd 2
sendSB sent FE0A26050108FFFF80971B01A3D856
readZB FE0166050062FE064180020202020702C2FE0166050062 rc=6
coordinator init cmd 3
sendSB sent FE032605870100A6
readZB FE0166050062FE064180020202020702C2FE0166050062FE0166050062 rc=6
coordinator init cmd 4
sendSB sent FE0426058302D8A3DD
readZB FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062 rc=6
coordinator init cmd 5
sendSB sent FE062605840400000100A4
readZB FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062 rc=6
coordinator init cmd 6
sendSB sent FE0D240014050F0001010002000015000020
readZB FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062FE0164000065 rc=6
coordinator init cmd 7
sendSB sent FE00260026
readZB FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062FE0164000065FE00660066FE0145C0088CFE0145C0088CFE0145C0088C rc=23
sendSB sent FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011 rc=19
found 0709, running oke
ZB coordinator started
trying pair inv 0
ecu_short = A3D8
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100703000319058FFFF10FFFF80971B01A3D8
sending paircmd 0
sendSB sent FE2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100703000319058FFFF10FFFF80971B01A3D8C2
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F0600703000319058
sending paircmd 1
sendSB sent FE1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F06007030003190588C
readZB FE0164020067FE1A448100000C02000014140124001891000000067030003190586A4C0DE9 rc=37
decoding : FE0164020067FE1A448100000C02000014140124001891000000067030003190586A4C0DE9
no valid pairing code, returning...
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000319058A3D810FFFF80971B01A3D8
sending paircmd 2
sendSB sent FE2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000319058A3D810FFFF80971B01A3D8BA
analyzing paircmd
readZB FE0164020067FE25448100000F010000141401210088A500000011703000319058A3D810FFFF80971B01A3D86A4C0D7D rc=48
decoding : FE0164020067FE25448100000F010000141401210088A500000011703000319058A3D810FFFF80971B01A3D86A4C0D7D
no valid pairing code, returning...
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D8
sending paircmd 3
sendSB sent FE1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
failed, inverter got id 0000
sending N.O. cmd
sendSB sent FE162401FFFF1414060001000F1E80971B01A3D8FBFB1100000D4F
readZB FE0164010064FE034480001401D2 rc=14
sendSB sent FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011 rc=19
found 0709, running oke
```

</details>

neruyzo, 2023-08-02:
> 

 `CC2530ZNP-with-SBL.hex` not `CC2530ZNP_2591-with-SBL.hex` from this folder ?

patience4711, 2023-08-02:
Yes, they both should work but i only can say for sure with the CC2530ZNP-with-SBL.hex.

BTW it looks like it should now but the A3D8 is the ECU short-id, which means that it didn't pair.

In the answers 1 or 2 the sought id should be right after the last occurance of the inverter serialnr, should not be the A3D8

neruyzo, 2023-08-02:
With `CC2530ZNP-with-SBL.hex` given by your link, the second inverter is "pair", the others fails with returning id 0000.

The poll is not working (the weather is really bad today, maybe some problem from this), when manually ask the first command `10;POLL=1` fail but the second return :

```text
zb send poll cmd inverter 1
--
sendSB sent FE1D24016A4C1414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE0F
readZB   FE0164010064FE034480001401D2FE0345C46A4C00A4FE0345C46A4C00A4FE0345C46A4C00A4FE7D4481000006016A4C1414001A000E0010000069703000734060FBFB5CBBBB20000500CBFFFF0000000080000AAA0902090400010000003900439ADD00000000FFFF0026074F000158DB000026BB0C01FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3573FEFE6A4C6AC6   rc=168
extracted sigQ = 10.2
decoding a DS3 inverter
extracted ACV = 15.0
extracted FREQ = 0.7
extracted HEATH = 13.2
* * * * polled inverter 1 * * * *
extracted time = 39645  the timespan = 278
* decoding panel 0   en_old 1.43
en_extr 88283.00  en_saved 1.47
en_incr 0.04  power 0.48
* decoding panel 1   en_old 0.16
en_extr 9915.00  en_saved 0.16
en_incr 0.00  power 0.00
no panel 2
no panel 3
total energy increase: 0.04
total energy stacked in Inv_Data[which].en_total: 1.63
```

[image]

The log for other doesn't look different than previous fail one (it's hard to capture the log from the web UI sorry).

<details>
<summary>End of log for pairing other inverter</summary>

```text
readZB FE0164020067  rc=6
--
decoding : FE0164020067
no valid pairing code, returning...
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000723550A3D810FFFF80971B01A3D8
sending paircmd 2
sendSB sent FE2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000723550A3D810FFFF80971B01A3D854
analyzing paircmd
readZB FE0164020067  rc=6
decoding : FE0164020067
no valid pairing code, returning...
pair command = 24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D8
sending paircmd 3
sendSB sent FE1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
failed, inverter got id 0000
sending N.O. cmd
sendSB sent FE162401FFFF1414060001000F1E80971B01A3D8FBFB1100000D4F
readZB FE0164010064FE034480001401D2  rc=14
sendSB sent FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011  rc=19
found 0709, running oke
```

</details>

neruyzo, 2023-08-02:
> Yes, they both should work but i only can say for sure with the CC2530ZNP-with-SBL.hex.
> 
> BTW it looks like it should now but the A3D8 is the ECU short-id, which means that it didn't pair.
> 
> In the answers 1 or 2 the sought id should be right after the last occurance of the inverter serialnr, should not be the A3D8
> 
> Where did you get the file you flashed on the zigbee?

I took the hex file from the ds3 folder, use the command below to convert to bin and flash from a windows computer using https://github.com/neruyzo/CCLoader.

```bash
objcopy --input-target=ihex --output-target=binary ESP-ECU_ADDON/CC25xxfirmware\ ds3/CC2530ZNP-with-SBL.hex cc2530_d0328.bin
```

patience4711, 2023-08-02:
@neruyzo  Well, it looks like it works oke. The software does what it is supposed to. The pairing is sometimes a bit difficult, that could also be a range problem.   I think it will be fine.

neruyzo, 2023-08-08:
Hi, I put an external global power meter on my full solar grid and every inverter/panel should work correctly.

I think the difficulty in pairing come from my panel/inverter configuration on floor, the panel 30 cm from ground and the distance of 15 cm between the panel/inverter make it hard to capture the signal (signal should have an hard time passing through the panel) .

Overall, thanks for confirming that every things work on software side and answer my questions. I close this issue because it work correctly with DS3-L.

closed 2023-08-08 by neruyzo