patience4711/read-APSystems-YC600-QS1-DS3 #138: Update the files

fwolfst, 2024-02-06:
I tried to compile the project, but I am quite sure that the source files are not up-to-date (the bin and main iso were updated often, but all the others were not). Could you upload the newest version?
I moved quite some files around and only noticed it just now (e.g. the LogEvent List name, domIdx variable etc.). Thanks for all that work!

patience4711, 2024-02-07:
@fwolfst Done. The file AAA_Decode has some changes as it turned out that the panels 1 and 2 of the yc600 where swapped. So this is new and not yet tested.

fwolfst, 2024-02-07:
> @fwolfst Done. The file AAA_Decode has some changes as it turned out that the panels 1 and 2 of the yc600 where swapped. So this is new and not yet tested.

Great. Thank you very much!

fwolfst, 2024-02-08:
> > @fwolfst Done. The file AAA_Decode has some changes as it turned out that the panels 1 and 2 of the yc600 where swapped. So this is new and not yet tested.
> 
> Great. Thank you very much!

Are you sure that also the main iso and the Logpage.ino are updated? Because in Logpage it tries to find Log_EventList. In the ESP-ECU-v10_9.ino I find Log_Events, which is probably the right variable, but different name.

patience4711, 2024-02-10:
@fwolfst  Yes, the files came from one directory. To be absolutely sure i uploaded them again. Sorry i have no time to investigate this but i did not have troubles with compiling so it should be oke.

fwolfst, 2024-02-10:
Much appreciated, thank you.

fwolfst, 2024-02-11:
Just FYI, I created a fork at https://github.com/fwolfst/ESP8266-APSystems . The goal is to make it compile on my machine without ArduinoIDE (because there the library management drives me nuts and under Linux the given code would not work). For me, platform.io is much closer to what I am used to from my dayjob.

I got it to compile but do not know if everything works (need again a working zigbee module + sun to try :) ). Also had to fix the WIFI settings, it wouldnt automatically reconnect after saving them on 192.168.4.1.

I will try to clean up the code (gives some warnings) and maybe (maybe maybe) integrate some lorawan if I win in the lottery-of-time. Tell me if you want to be pinged with updates. Unfortunately, github PRs seem not to work best in this setting.

joffito, 2024-02-11:
That sounds great! Maybe you can even fork the ESP32 version? But first things first. Hope you can get your hands on a zigbee module. if not i could order one and flash it when i got it.

patience4711, 2024-02-12:
The wifi connection portal is sometimes not working 100% probably because during the connection attempt, there are changes in wifi mode. This can lead to disconnection with the client that is configuring.  

I think it would even be better to abandon the esp8266  version and switch to ESP32.  
It is not more expensive and has the possiblity to connect wifi via bluetooth . This works very reliable and saves much programspace and some global vars. With our phone we can have a terminal to the esp so we could even port the console to that.  It would even be possible to have the console via BT. The wifiportal and the console are massive pieces of code so if this could be slimmed down....

gismo2004, 2024-03-04:
@patience4711 if you think about "moving" something, just check out esphome --> basically esphome has everything available when it comes to connection/hompage/mqtt and things like that, so you just have to care about the core thing, which is the communication part.

djsys, 2024-03-05:
Je viens de la version pour ESP8266, Je viens d'essayer la version 8 pour ESP32, impossible de faire "PAIR" plus de 20 tentatives sans succes et la console se déconn
[image]
ecte après chaque commande ?

closed 2024-03-26 by patience4711