patience4711/read-APSystems-YC600-QS1-DS3 #139: Flash Firmware to CC2530

FreeTechNick, 2024-03-04:
Hi,

I'm searching for a for me understandable description how to flash the CC2530 via ESP8266. I have an CC2530 with 24pins and don't know how to connect the CC2530 to ESP8266 correct for flashing. Also a description for flashing would be helpful. 

Thank you.

Markus
[image]

swbouman, 2024-03-04:
Hey Markus, 

thank you for the message. Please check the hardware chapter in the wiki. Here is described how to connect the zigbee radio to a esp8266 to flash it with the correct firmware.

FreeTechNick, 2024-03-04:
Hi @swbouman,

thanx I have seen the wiki chapter but this does not match with my version of the cc2530  (PIN count and labels). Is there a mapping table?

joffito, 2024-03-04:
i think i have the same module like you. basically the P23 and P24 is missing. Did you try using  P21 & P22 as well as P02 & P03 and the reset line? i don't remember how i did it but i wasn't able to kill my Cc2530 ;)

swbouman, 2024-03-04:
[image]

swbouman, 2024-03-04:
Just copy the wiring as shown on the picture. Follow the red lines for flashing.

joffito, 2024-03-04:
His module only shows 12 pins on the right whereaa the picture of the module in the wiki shows 14.

patience4711, 2024-03-05:
The drawing shows pin 2_2 (22) (dc) and 2_1 (21) (dd) to use for flashing. So if you have a module with 12 14 or 88 pins, just use 22 and 21.

closed 2024-03-27 by patience4711