patience4711/read-APSystems-YC600-QS1-DS3 #141: Possibility to pair an ECU-B and this ECU ?

sebastianrothe, 2024-04-12:
Hey, thanks for the awesome project. 

I would like to know if it is possible to pair multiple ECUs(this one **AND** the ECU-B) to one inverter? I need the original ECU-B from APSystems connected for extended warranty.

The other one always seems to lose the connection once I pair one of them.

togoright, 2024-04-13:
You can try to enter the ID from your ECU-B into the ESP-ECU. Then both should work at the same time.

sebastianrothe, 2024-04-13:
Thanks, I will give it a try.

sebastianrothe, 2024-04-13:
Unfortunately, I can't get it to work with both ECUs. It is always working only for the one, I paired last.

patience4711, 2024-05-08:
In this zigbee network there is only one coodinator where the inververter is paired to. So no...

closed 2024-05-08 by patience4711