patience4711/read-APSystems-YC600-QS1-DS3 #143: Adding Zigbee router

ALaDoffe, 2024-06-04:
Hi 

Thx for your great job.
Is there a way to pair additional zigbee router to the coordinator in order to extend range between esp-ecu and the APS ?
My APS is quite far from the esp-ecu.

Regards

patience4711, 2024-06-10:
In theory every node in the network should work as a repeater (if i understand it well). Sadly our system only paires with the APS devices. I know APS sell a zigbee booster you could connect to the antenna adapter of your zigbee device but if this will work? 

I assume that you use the zigbee module with the  CC2591 amplifier.  I would say place your ESP-EVU closer to the inverters and use a wifi repeater. 
I am sorry, i have no other helpfull suggestions.

ALaDoffe, 2024-06-19:
OK OK thx for your response.
As a workaround, I changed my CC2530 antenna.
The signal is now quite good (around 25%)

closed 2024-06-19 by ALaDoffe