patience4711/read-APSystems-YC600-QS1-DS3 #144: Error adding the ninth inverter

pdufrenne, 2024-07-30:
High,

I appreciate the effort you put into this software.
However, I encountered a problem when added the ninth inverter. (I have exactly nine inverters)
The Add button and all other options are gone.
See the picture.

[image]

I use a Wemos D1 Mini and a CC2530 Zigbee module

Thanks in advance,

Peter

patience4711, 2024-08-02:
Hi Peter, i remember we had this problem with the esp32 version, that was a problem in the html of the inverterpage. 
So i made some changes in the html and compiled a new version.  Find it [here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU-v10_9a.bin). Sadly i have no means of testing so i cannot guarantee that it will work.
If it doesn't work i would like to suggest that you buy an ESP32 which would probably work better with 9 inverters.. Please look [here](https://github.com/patience4711/ESP32-read-APS-inverters)

patience4711, 2024-08-09:
No thanks, you're welcome

closed 2024-08-09 by patience4711