patience4711/read-APSystems-YC600-QS1-DS3 #145: Zero inyection

JMMarquezL, 2024-08-05:
I'm specifically interested in whether it's feasible to control the microinverter's zero injection mode via the ESP ECU. Thanks.

patience4711, 2024-08-07:
If you know the zigbee command it would be doable via the console. Sadly i don't know the command.

JMMarquezL, 2024-08-07:
How can I acquire the command? What was your method for decoding the Zigbee data? Thank you very much.

patience4711, 2024-08-09:
This question has been asked before.
read [here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/118) and
to get an idea how the reversed-engineering went, read [here](https://github.com/Koenkk/zigbee2mqtt)

In short: Someone with an official ECU needs to sniff the communication ecu<->inverter.

closed 2024-09-04 by patience4711