patience4711/read-APSystems-YC600-QS1-DS3 #147: Impact of 30-second polling interval on zigbee stability

Adamou02, 2025-03-06:
Hi,

I understand that high polling rates can cause stability issues in the Zigbee module, but I would like to monitor my solar panel production in real time to compare it with my home network consumption.

I am using a Shelly device to measure my total home consumption at the main entry point of my network. To accurately calculate the energy balance, I need my solar production readings to be at the same frequency as my Shelly measurements.

Would setting the polling interval to 30 seconds cause significant stability issues in the Zigbee module? Or is this frequency still acceptable?

Thank you for your help!

joffito, 2025-03-06:
Even AP Systems uses a 5 min polling intervall. i think that shows how fragile the system is. but from the past i think a more frequent interval always caused stability issues with the esp.

togoright, 2025-03-06:
Please have a look [here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/97).

Since the test with the 10-second interval for two days, I've been polling every two minutes until today without any problems. I assume that every 30 seconds shouldn't cause any problems either.

mi-hol, 2025-03-06:
> Even AP Systems uses a 5 min polling intervall. i think that shows how fragile the system is

@joffito could you provide measured facts supporting  this opinion?

Even a 5 seconds polling intervall should work, under the assumption that polling is only performed when inverter is connected to ECU. The inverter disconnects at sundown and reconnects at sunup. 
The actual time for these events changes by day and location!

patience4711, 2025-03-11:
The stability problem occurred when the zigbee healthcheck and the polling are done at the same time. The software prevents this. When you poll every 5 seconds this would probably happen a lot. 
But every 30 secs...., there is only one way to find out. Disable the automatic polling and send a http or mqtt polling request (see wiki) every 30 seconds. With not too many inverters it may work.

closed 2025-03-31 by patience4711