patience4711/read-APSystems-YC600-QS1-DS3 #149: Request data via http: total energy (eN)

markvd1962, 2025-03-28:
I am a very happy user with YC600's, QS1's and DS3 connected. I am pulling data via http into (and from) Home Assistant. It works fine. Only the "total energy" (called eN in the json message) shows a bit strange behaviour: Unfortunately it measures on a daily basis, but it seems it resets to zero at sunrise (when the inverters kick in) rather than at midnight. So I have difficulties loading the total energy in the HA energy dashboard. Anybody found a solution? After about 3 evenings (3x4 hours) trying to find a solution I found no solution (but yes, "trying" is part of the fun).
PS: I don't think the "pulling data via http" in the Wiki (miscellaneous section) is correct anymore.

markvd1962, 2025-03-31:
What is incorrect in the wiki:
You say "In the json total energy is en_total, meaning the total of the specific inverter". But in the wiki (08 miscellaneous) it mentions ""en":[1081.06,1064.31,"n/e","n/e"]}", so total energy per panel, not per inverter. I saw the total enery is per panel.

The energy per solar panel is not reset:
Are you sure? I saw the total enery was reset to zero automatically when the inverters started working again at sunrise.

patience4711, 2025-03-31:
I meant at midnight the "en" (energy) values are not reset, the "en_total" is. The "en" value shows only the energy produced (per panel) so far on the current day. The en_total is reset on demand of users who like to start at null at midnight. I think when you don't collect data whe there isn't any (during darkness) there can't be a problem. 
Anyway you saw the confusion now so i hope the issue can be closed?.

markvd1962, 2025-03-31:
Thanks for the detailed explanation. And again I learned a lot. Yes, issue can be closed.

closed 2025-04-01 by patience4711