patience4711/read-APSystems-YC600-QS1-DS3 #150: Write max power limit parameter to DS3-H microinverters via Zigbee

seby90, 2025-04-02:
Hello, and thank you for your excellent work on reverse-engineering the Zigbee communication protocol used by APSystems microinverters.

I would like to know if it would be possible to add a feature allowing to write parameters directly to the microinverters, specifically to modify the maximum power limit and thereby bypass the restriction currently enforced by the ECU-C and/or the EMA platform.

Context:
Following a recent update of the EMA platform, APSystems has reinforced a remote limitation applied to DS3-H microinverters.
Previously, it was possible to set a maximum power limit higher than 500 W per microinverter (e.g., 960 W for a DS3-H) from the EMA interface, and the microinverters would actually deliver this power.
Since this update, even if 960 W is configured in EMA, after approximately 10 minutes, the EMA server automatically sends a Zigbee command to force the power limit down to 505 W per microinverter.

Additionally, it is possible to adjust the maximum power limit locally via the hidden menu of the ECU-C. However, this interface restricts the input to 500 W maximum, preventing any customization beyond that limit.

Since these limitations have been applied, I have noticed that my DS3-H microinverters no longer exceed 920 W of instantaneous maximum power each, whereas previously they regularly reached 970 W.
When multiplied by the number of inverters in the installation, this limitation results in a significant loss of instantaneous production capacity.

Another important point:
I have also identified a bug in the behavior of the ECU-C when using the Zero Export function.
After enabling this function, if I later disable it, the maximum power limit of the microinverters remains constrained.
It is then necessary to manually reconfigure the maximum power limit via the ECU-C interface; otherwise, the microinverters continue to operate with a power restriction, even after Zero Export has been disabled.

Request:
I would like to know if it would be technically possible to implement support for sending a Zigbee write command to adjust the maximum power limit parameter directly within the microinverters, and thus remove the current restriction.

I am willing to assist with testing this feature and, if needed, to provide Zigbee traffic captures observed during EMA-initiated power limit changes.

Thank you in advance for your feedback.

patience4711, 2025-04-25:
It is possible to send a zigbee command, this is already posible through the console now. If one knows what to send. The zigbee trafic captures are necessary for that. If you have that, you can test yourself, I can't because i don't have this inverter. 
So yes, technically it is possible if there is enough programming space, otherwise the esp32 has it.

seby90, 2025-04-26:
So all you need to do is capture the data that the ECU-C sends to the micro-inverters to increase the power? And the data is sent in clear text without any encryption?

Is it possible with your receiver to capture the data that the ECU sends to the micro-inverters?

Thank you in advance for your feedback.

patience4711, 2025-05-01:
The data is send as a string of hexadecimal characters. You need a zigbee sniffer to capture the data.

seby90, 2025-05-07:
Ok for the sniffer , i must try.

I have built and tested an ECU-ESP32 with my three DS3-H units. These three DS3-H micro-inverters were originally connected to and managed by an APSystems ECU-C. To get the ESP32 to connect to all three micro-inverters, I had to disconnect the mains power, then perform a pairing; only after that could I see and receive the Zigbee data from the micro-inverters. The problem now is that my ECU-C doesn’t see anything at all anymore. As a result, I had to remove the three micro-inverters from my ESP32, cut the power again, and then resynchronize with my ECU-C.

My ultimate goal is to be able to send power-update commands to the micro-inverters via Zigbee while still keeping the main ECU-C connected. Is this possible?

seby90, 2025-05-17:
I’ve received my CC2531 sniffer dongle (flash with Zboss Sniffer). I’m capturing frames on Channel 16 labeled “APS,” and in some frames under the DATA field I can see the serial numbers of the micro-inverters as well as other data that I don’t know how to decode. Do you know of any site that explains how to decode these frames?

Next, I’d like to capture the data sent from the ECU-C to the micro-inverters in order to modify their maximum power. Which frames should I send you so that you can potentially update your application to change the micro-inverters’ max power?

seby90, 2025-06-07:
Hi @patience4711 , I'm trying to send a command to set the maximum output power of a DS3-H microinverter to 100 W.
Based on previously sniffed traffic between the ECU-C (0x0000) and the inverter (0x1822), we identified the relevant frame for 100 W:

Original (sniffed) frame:
61888fc697221800004800221800000fb000140600050f140580971b0597c6fbfb06aa270000067b0158fefe

This was then reversed byte-by-byte, as Zigbee frames in the sniffer appear inverted:

FEFE58017B06000027AA06FBFBC697051B978005140F0500061400B00F0000182200480000182297C68F8861

As documented in the [Testing and Debugging guide](https://github.com/patience4711/read-APSystems-YC600-QS1-DS/wiki/09-TESTING-AND-DEBUGGING), the console command zbt= only requires:
The raw payload (command + data),
Without the leading FE,
And without the final checksum byte (which is calculated by the firmware).

10;zbt=58017B06000027AA06FBFBC697051B978005140F0500061400B00F0000182200480000182297C68F88

The console prints:

the command = 58017B06000027AA06FBFBC697051B978005140F0500061400B00
sendZB FE1858017B06000027AA06FBFBC697051B978005140F0500061400B0056
readZB nothing to read

The frame appears truncated at ...B00, before reaching the 0F00001822... portion.
I’ve also sniffed the Zigbee traffic while sending this command and I see nothing leaving the ESP32.

A bit of help would be truly appreciated.

swbouman, 2025-06-13:
I sended the command with my inverter code and my Ecu code. But then 3 inverters went crazy. I saw 1 panel of more than 10000 watts! After a minute the panels went back to normal state. So I believe something went wrong with the code.

seby90, 2025-06-13:
@swbouman Thank you for your feedback. I’d really like to know the exact frame you sent — was it 10;ZBT=... ? Could you share the full command you used?

swbouman, 2025-06-13:
My pleasure! I sended indeed the 10;ZBT= command with your string and replaced the inverter and Ecu ids with mine. I’ve 7 inverters DS3 and sended the command to just 1 inverter. But 3 went crazy.

seby90, 2025-06-13:
@swbouman Did you try sniffing at the same time as sending the 10;ZBT= command? Because in my case, no matter what I send, I don’t see any frame appearing in Wireshark — so for now, I haven’t been able to send any frame successfully.

patience4711, 2025-07-09:
I see a lot of interesting activity here after a long summer holliday.  I am affraid i cant help much, If sending via the console won't work, try serial 10;ZBT=

I am looking at the code now but i have to setup an ESP32ECU to see if i can reproduce this behaviour. But i really have no time to do it at the moment.

kadzsol, 2025-07-12:
Nice project. Here is some information that might help:

Please share full traces instead of just some messages. APS might have used a sequence of messages that are relevant for this use case. I recommend at least the following use cases to sniff:

1. normal operation (no additional commands, just let your system run, preferrably with only 1 inverter).
2. adjust power to a value (1 inverter)
3. adjust power to another value (1 inverter)

Use TI's SmartRF packet sniffer (google SWRU187G) instead of Wireshark. In my experience, this is much easier to read/interpret as it was designed for this kind of applications.

Study/Understand the Zigbee protocol. APS did not create a standard Zegbee application but does use the protocol layer. Understand for instance the difference between Zegbee broadcast and single messages. I found this document particularly helpful in the past: https://studylib.net/doc/18864951/developing-a-zigbee-system-using-cc2530

Use Z-tool to experiment with sending commands (part of https://www.ti.com/tool/Z-STACK) to avoid any possible errors by using a non-standard tool.

Sniff what you send to confirm your message out is the one you intended.