patience4711/read-APSystems-YC600-QS1-DS3 #151: Sniffing

jaydouble, 2025-04-08:
Hi,

I own a ECU-R-Pro, and 9 inverters, and 18 solar panels.

I can sniff how things like on/off, and limiting works, but... I don't know how to sniff.

I don't have an ESP ECU (yet). maybe i'll get someday a CC2530 (or CC2531). 

Is there a great how to sniff zigbee for noobs? 
And what kind of devices do I need for sniffing zigbee traffic?

swbouman, 2025-04-08:
I’m also interested in a bigger code, so we could probably replace the ecu-r for this project. Would be wonderful. 

I don’t own an ecu-r so I’m not able to sniff. 

First good option would be to change the gridprofile to avoid blackouts during the day due to higher voltages on the mains here. And possibility to manage frequency shifting when using an external battery and avoid net inserts. 

Maybe software updates of the inverters but not real main issue at the moment. 

I believe the esp and esp32 should be able to manage these options. 

For the sniffing we need an ecu-r and what I’ve been reading an esp with zigbee module. Then send the instruction to the inverters, get the code and implement.

patience4711, 2025-04-12:
I wouldn't know about zigbee sniffing for noobs but there is enough to be found on the internet.
I once built a sniffer but that was a long time ago and I don't remember exactly anymore. I had a cc2531 flashed with sniffer firmware (this is the difficult part because you have to connect it to an esp) and a laptop to plug it in, with wireshark and another software i forgot.

[here](https://github.com/Koenkk/zigbee2mqtt/issues/4221) is more to find about setting up a sniffer and how to do it, just ctrl-F sniffing.

I think there are a lot of people interested in these sniffs to extent the capabilities of the esp-ecu.